ros-industrial-attic/robotiq #24: Robotiq_c2_model visualization in gazebo doesn't spawn fully

TheDash, 2015-03-04:
Only the base of the hand is there (e.g not the 2 fingers that move inward). However, in Rviz, i can fully see the model.

TheDash, 2015-03-23:
ping @shaun-edwards 

The issue here is that there is no inertial elements for the c2 model gripper. IS it possible we can get some good values for this so I can insert it into the model. This will enable Gazebo capability.

shaun-edwards, 2015-03-23:
We had some discussion about utilizing the values for the model c gripper, see #16.  Ultimately, we pulled them out because we weren't sure how accurate they were.  We asked @nicolaslauzier, to provide accurate numbers in the future, but without anybody asking, there was no urgency.  @nicolaslauzier, could you provide the c2 inertia values, sooner rather than later.

TheDash, 2015-03-24:
Thanks. That should get the ball rolling on this.

nicolaslauzier, 2015-03-24:
I'll bring it to our internal discussions but I think it will be difficult to make it a short term priority. Should we also provide data regarding the spring for modeling the under-actuation?

TheDash, 2015-03-24:
That would be ideal. We are planning to use it sometime end of May (ICRA 2015) and want to have the simulation up and ready before the real robot gets built again. I am hoping to have the simulation done either this week or next week. The limiting factor is not being able to use the proper gripper, so I'm using the PR2's gripper right now. 

Whenever you guys choose to get it done, it's up to you. But sooner is better. The spring modelling is up to you as well, I'm not experienced enough to know if that will make that much of a difference from the real model.

nicolaslauzier, 2015-03-25:
Just curious, is it to be used for the Amazon picking challenge?

I think that having the proper spring data is essential to model the gripper correctly but I haven't tried the 3F gripper model so I cannot tell for sure.

A lot of people are away this week so I should be able to provide you an answer early next week.

TheDash, 2015-03-25:
It will just be at ICRA for demos.

J-Schaefer, 2015-11-17:
Did someone solve the problem right now?

I tried to spawn the robotiq_c2_gripper attached to an UR5 in Gazebo, but only the base of the gripper was shown. With this i got the message ```Assimp reports no scene in package:// ... ```
(When i extract meshes from the Step-File of the newer 2-Finger-85 and load them into the model, i no longer get this message, but the gripper will still not spawn completely)

I also wanted to spawn the model as a stand-alone-model with:

```rosrun gazebo_ros spawn_model -file robotiq_c2_model.xacro -urdf -x 0 -y 0 -z 0 -model robotiq_c2_model```

and got the error:

```cpp
Error [parser_urdf.cc:2608] Unable to call parseURDF on robot model
Error [parser.cc:299] parse as old deprecated model file failed.
Error [World.cc:1527] Unable to read sdf string[<!-- <robot name="robotiq_c2_model" xmlns:xacro="http://ros.org/wiki/xacro"> --><robot xmlns:xacro="http://www.ros.org/wiki/xacro" name="robotiq_c2_model"><xacro:macro name="robotiq_c2_model" params="prefix"><xacro:include filename="$(find robotiq_c2_model_visualization)/urdf/robotiq_c2_model_macro.xacro" /><xacro:robotiq_c2_model prefix="" /></xacro:macro><origin xyz="0 0 0" rpy="0 -0 0" /></robot>]```

Another thing that i tried, is to set all joints to 'fixed'. The gripper now spawns completely, but that is not a practicable solution because the gripper should be able to move. 

Did someone find a solution? And what about those inertial elements?

J-Schaefer, 2015-12-02:
I found a model of the Robotiq 85 Gripper created by [StanleyInnovations](https://github.com/StanleyInnovation/robotiq_85_gripper).
It is not the C2 Gripper but it works quite fine. And there are inertial values given.