ros-industrial-attic/robotiq #26: catkin_lint errors

gavanderhoorn, 2015-05-14:
```
robotiq_c_model_control: CMakeLists.txt(8): error: variable EXECUTABLE_OUTPUT_PATH is modified
robotiq_c_model_control: CMakeLists.txt(10): error: variable LIBRARY_OUTPUT_PATH is modified
robotiq_c_model_control: notice: package description starts with boilerplate 'Package to'
robotiq_force_torque_sensor: error: message dependency 'std_msgs' is not listed in catkin_package()
robotiq_force_torque_sensor: warning: package 'std_msgs' should be listed in catkin_package()
robotiq_force_torque_sensor: notice: package description starts with boilerplate 'Package for'
robotiq_force_torque_sensor: notice: target 'rq_sensor' is not installed
robotiq_modbus_rtu: CMakeLists.txt(7): error: variable EXECUTABLE_OUTPUT_PATH is modified
robotiq_modbus_rtu: CMakeLists.txt(9): error: variable LIBRARY_OUTPUT_PATH is modified
robotiq_modbus_tcp: CMakeLists.txt(7): error: variable EXECUTABLE_OUTPUT_PATH is modified
robotiq_modbus_tcp: CMakeLists.txt(9): error: variable LIBRARY_OUTPUT_PATH is modified
robotiq_s_model_control: error: unconfigured build_depend on 'message_generation'
robotiq_s_model_control: error: unconfigured build_depend on 'rospy'
robotiq_s_model_control: error: unconfigured build_depend on 'message_generation'
robotiq_s_model_control: CMakeLists.txt(4): error: missing COMPONENTS keyword before 'rospy'
robotiq_s_model_control: CMakeLists.txt(4): error: missing COMPONENTS keyword before 'message_generation'
robotiq_s_model_control: CMakeLists.txt(7): error: variable EXECUTABLE_OUTPUT_PATH is modified
robotiq_s_model_control: CMakeLists.txt(9): error: variable LIBRARY_OUTPUT_PATH is modified
robotiq_s_model_control: notice: package description starts with boilerplate 'Package to'
robotiq_s_model_visualization: error: unconfigured build_depend on 'urdf'
robotiq_s_model_visualization: CMakeLists.txt(4): error: missing COMPONENTS keyword before 'urdf'
```

gavanderhoorn, 2015-08-04:
Update (`catkin_lint -W2 robotiq` run against repo at f3e1c987):

```
robotiq_action_server: error: missing build_depend on 'message_generation'
robotiq_action_server: error: missing build_depend on 'control_msgs'
robotiq_action_server: CMakeLists.txt(34): notice: line is not indented properly
robotiq_action_server: CMakeLists.txt(47): notice: line is not indented properly
robotiq_action_server: CMakeLists.txt(58): notice: line is not indented properly
robotiq_c_model_control: error: catkin_package() exports package include path that is not installed
robotiq_c_model_control: CMakeLists.txt(8): error: variable EXECUTABLE_OUTPUT_PATH is modified
robotiq_c_model_control: CMakeLists.txt(10): error: variable LIBRARY_OUTPUT_PATH is modified
robotiq_c_model_control: warning: target 'c_model_ethercat_node' is not installed
robotiq_c_model_control: notice: package description starts with boilerplate 'Package to'
robotiq_ethercat: error: unknown run_depend 'soem'
robotiq_ethercat: error: unknown build_depend 'soem'
robotiq_ethercat: CMakeLists.txt(4): error: unknown package 'soem'
robotiq_ethercat: CMakeLists.txt(9): error: unknown package 'soem'
robotiq_force_torque_sensor: error: message dependency 'std_msgs' is not listed in catkin_package()
robotiq_force_torque_sensor: warning: package 'std_msgs' should be listed in catkin_package()
robotiq_force_torque_sensor: notice: package description starts with boilerplate 'Package for'
robotiq_force_torque_sensor: notice: target 'rq_sensor' is not installed
robotiq_modbus_rtu: CMakeLists.txt(7): error: variable EXECUTABLE_OUTPUT_PATH is modified
robotiq_modbus_rtu: CMakeLists.txt(9): error: variable LIBRARY_OUTPUT_PATH is modified
robotiq_modbus_tcp: CMakeLists.txt(7): error: variable EXECUTABLE_OUTPUT_PATH is modified
robotiq_modbus_tcp: CMakeLists.txt(9): error: variable LIBRARY_OUTPUT_PATH is modified
robotiq_s_model_control: error: unconfigured build_depend on 'message_generation'
robotiq_s_model_control: error: unconfigured build_depend on 'robotiq_ethercat'
robotiq_s_model_control: error: unconfigured build_depend on 'rospy'
robotiq_s_model_control: error: unconfigured build_depend on 'roscpp'
robotiq_s_model_control: error: unconfigured build_depend on 'message_generation'
robotiq_s_model_control: CMakeLists.txt(4): error: missing COMPONENTS keyword before 'rospy'
robotiq_s_model_control: CMakeLists.txt(4): error: missing COMPONENTS keyword before 'message_generation'
robotiq_s_model_control: CMakeLists.txt(4): error: missing COMPONENTS keyword before 'roscpp'
robotiq_s_model_control: CMakeLists.txt(4): error: missing COMPONENTS keyword before 'robotiq_ethercat'
robotiq_s_model_control: CMakeLists.txt(7): error: variable EXECUTABLE_OUTPUT_PATH is modified
robotiq_s_model_control: CMakeLists.txt(9): error: variable LIBRARY_OUTPUT_PATH is modified
robotiq_s_model_control: warning: target 's_model_ethercat_node' is not installed
robotiq_s_model_control: notice: package description starts with boilerplate 'Package to'
robotiq_s_model_visualization: error: unconfigured build_depend on 'urdf'
robotiq_s_model_visualization: CMakeLists.txt(4): error: missing COMPONENTS keyword before 'urdf'
catkin_lint: checked 10 packages and found 37 problems
```

gavanderhoorn, 2020-07-07:
@ipa-nhg: I would recommend to re-run `catkin_lint`.

It's been a while since I reported this (almost 5 years ..) so lots of things have changed in the meantime.

ipa-nhg, 2020-07-07:
Sure, now catkin_lint is reporting even more errors...

ipa-nhg, 2020-07-07:
Solved by https://github.com/ros-industrial/robotiq/pull/176

gavanderhoorn, 2020-07-07:
@ipa-nhg: CI seems to fail:

```
CMake Warning (dev) at CMakeLists.txt:54 (add_dependencies):
  Policy CMP0046 is not set: Error on non-existent dependency in
  add_dependencies.  Run "cmake --help-policy CMP0046" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.
  The dependency target "robotiq_ft_sensor_gencfg" of target "rq_test_sensor"
  does not exist.
This warning is for project developers.  Use -Wno-dev to suppress it.
```

ipa-nhg, 2020-07-07:
@gavanderhoorn sorry, I made a typo mistake -> solved and rebased the branch, now should be ok

gavanderhoorn, 2020-07-07:
Just left some comments.