ros-industrial-attic/robotiq #37: Improper clean up of serial interface on ctrl-c of rq_sensor

futureneer, 2015-07-06:
Hello,
When ctrl-c is invoked on rq_sensor, I believe the serial interface is not being cleaned up properly. I think this because when the sensor is restarted shortly after the ctrl-c, it takes MUCH longer to start streaming at full speed. If you wait a few minutes after invoking ctrl-c, it starts just fine.  I have had this same behavior with serial connections that are not closed properly, so that is my gut instinct.  I will be looking through the code, but in the mean time I thought I would let you know.

Cheers,
Kel

nicolaslauzier, 2015-07-06:
I don't know if there is a way to catch ctrl+c inside of a ROS node. If so, it would make sense to close the serial port properly at this moment.

As far as I recall, there is no such mechanism implemented in the node for the FT sensor.

shaun-edwards, 2015-07-06:
I believe this can be done with a custom SIGINT handler, see [here](http://wiki.ros.org/roscpp/Overview/Initialization%20and%20Shutdown#Custom_SIGINT_Handler)

futureneer, 2015-07-07:
I mean, couldn't you just put some sort of clean up command after the
while(ros::ok) loop that closes everything down? That seems to be a common
pattern.