ros-industrial-attic/robotiq #38: EtherCAT robotiq packages require external library

shaun-edwards, 2015-07-07:
The EtherCAT packages require a forked version of [SOEM](https://github.com/ros-industrial/ethercat-soem).  This is not documented anywhere and could potentially trip up some users.

A *Installation from Source* section should be added to the [wiki page](http://wiki.ros.org/robotiq).  Current dependencies are:
 - SOEM
 - pyModbus - From tutorial `Finally, make sure that the external dependency for the package "robotiq_modbus_tcp" has been installed. The dependency is the python package pyModbus. On Ubuntu Precise, it is simply installed using "rosdep install robotiq_modbus_tcp". On other systems, it can be installed using "easy_install -U pymodbus"`

shaun-edwards, 2015-07-07:
@Jmeyer1292, I haven't verified this issue, but I believe we need to add these instructions to assist the users in installing dependencies.  This should be a quick change.

gavanderhoorn, 2015-07-07:
Please avoid using `easy_install` and / or `setuptools`, use `pip`. But then only if really necessary, as it circumvents your system's package manager.

Jmeyer1292, 2015-07-07:
I'll get on it.

shaun-edwards, 2015-07-08:
I added installation instructions to the [wiki](http://wiki.ros.org/robotiq#Installation).  This is just a temporary fix until #39 is addressed.

closed 2015-07-08 by shaun-edwards

theNerd247, 2015-09-11:
The documentation on the wiki is a bit misleading. It requests the package [here][1] to be installed in the catkin workspace however, robotiq_ethercat requires  the soem package...it's causing a bunch of catkin_make errors. Namely, the soem package can't be  found. According to #39 I'm assuming we should go for the debian package instead? Or at least that's what the robotiq_ethercat package is assuming....

[1]: https://github.com/ros-industrial/ethercat-soem

Jmeyer1292, 2015-09-11:
Yes, I believe since the change you should use the debian package for soem that can be obtained by ```sudo apt-get install ros-indigo-soem```. I will update the wiki.

theNerd247, 2015-09-11:
On [this][1] page. The section that says to clone this repo: 

    git clone https://github.com/ros-industrial/ethercat-soem.git

I believe

    sudo apt-get install ros-indigo-soem

Is what you mean now - or at least that's what worked for me.

[1]: http://wiki.ros.org/robotiq

shaun-edwards, 2015-09-11:
@theNerd247, my apologies.  I fixed the original issue, but failed to update the documentation.  I've update the wiki with the apt-get installation.  Thanks for pointing this out!