ros-industrial-attic/robotiq #39: Utilize soem ROS package instead of forked ethercat_soem

shaun-edwards, 2015-07-07:
A ROS SOEM library already exists as a debian in Indigo (it's built from [here](https://github.com/smits/soem)).  The latest merges to support an EtherCAT version of the robotiq gripper require a forked version, [here](https://github.com/ros-industrial/ethercat-soem).  

I believe the original version was forked to provide catkin support.  This now exists in the original SOEM version.  If we utilized the existing ROS debian version of SOEM, we could do official debian releases of the robotiq packages.

shaun-edwards, 2015-07-07:
@Jmeyer1292, can you comment?

Jmeyer1292, 2015-07-07:
@shaun-edwards I'm not attached to my version of SOEM. I'll test this one out with our drivers and report back.

shaun-edwards, 2015-07-11:
I started a fix [here](https://github.com/shaun-edwards/robotiq/tree/issue39_soem_lib), but I'm blocked until smits/soem#4 is addressed.

closed 2019-01-25 by jproberge