ros-industrial-attic/robotiq #42: Control msg dependency not properly declared

shaun-edwards, 2015-08-04:
The `robotiq_action_server` references `control_msgs` in the CMakeLists.txt, but it is not in the package.xml.

This appears to be causing the following [failure](http://jenkins.ros.org/job/devel-indigo-robotiq/ARCH_PARAM=amd64,UBUNTU_PARAM=trusty,label=devel/9/console) on the build farm.

shaun-edwards, 2015-08-04:
I wonder if `catkin_lint` would have caught this issue?

gavanderhoorn, 2015-08-04:
yep:

```
robotiq_action_server: error: missing build_depend on 'message_generation'
robotiq_action_server: error: missing build_depend on 'control_msgs'
robotiq_action_server: CMakeLists.txt(34): notice: line is not indented properly
robotiq_action_server: CMakeLists.txt(47): notice: line is not indented properly
robotiq_action_server: CMakeLists.txt(58): notice: line is not indented properly
```

gavanderhoorn, 2015-08-04:
See also #26.

closed 2018-07-19 by jproberge