ros-industrial-attic/robotiq #44: Better UI for Robitq SModel

theNerd247, 2015-09-15:
I may be unaware of this, however the user API for the SModel seems a bit raw. What I mean is, is there an API such that I don't have to deal with the registers on the gripper directly. The robotiq_s_model_controller/SModelTcpNode seems to act as a wrapper with a transparent interface - rather than an abstraction. For example it would be nice to have an API that deals directly with the SModelTcpNode under the hood (i.e. subscribing and publishing) while allowing the user to send commands like (`setMode(...)` or `resetGripper(...)`). This way I won't have to write my own code that checks to make sure the gripper has been properly initialized and activated and in the correct mode. 

I did find the robotiq_action_server node however it appears to only be for C models. If not are their any docs on how to use this?

fevb, 2015-09-15:
Hi,

I have developed a bit of a C++ / python wrapper for the S model:
https://github.com/kth-ros-pkg/robotiq_s_model_utils/tree/hydro/robotiq_s_model_control_client

You can see an example of the python wrapper in the **scripts** folder

theNerd247, 2015-09-15:
Great! I'll take a look at it. I'm using c++ for development.

theNerd247, 2015-09-15:
So I think the library is a good start. I think it'll need some cleanup however it would be nice to see it merged with this package.

closed 2019-04-30 by jproberge