ros-industrial-attic/robotiq #45: Can not connect gripper by USB

james2254, 2016-01-11:
When I run  rosrun robotiq_c_model_control CModelRtuNode.py,there is a misstake.What should I do?
[image]

shaun-edwards, 2016-01-11:
@james2254, looks like the node is expecting command line arguments.

james2254, 2016-01-12:
@shaun-edwards I download the package and run the node by the ROS wiki .Change the node or other   
method, which can solve this issue ?  thank you !

carlosjoserg, 2016-08-01:
Hi @james2254 , don't know if you solved the issue.

After you connect the gripper, check with `lsusb` it connected, you should see something like:

`Bus 001 Device 008: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)`

Then check with `dmesg | grep tty` what port it connected to. You should see something like:

`[ 6416.940532] usb 1-2: FTDI USB Serial Device converter now attached to ttyUSB0`

If that all goes well, then you also need to create a modbus config file (at least for me, it didn't work until I did this), you can run this bash script to do so:

``` bash
# configuration of virtual port com FT X series
FTDI_CONF_FILE='/etc/modprobe.d/ftdi_sio.conf'
FTDI_CONF_OPTION='options ftdi_sio product=0x6015 vendor=0x0403'
FTDI_CONF_ALIAS='alias usb:v0403p6015d*dc*dsc*dp*ic*isc*ip* ftdi_sio'
if [ -f $FTDI_CONF_FILE ]; then
        MSG="FTDI conf file already exists."
        echo $MSG
else
        MSG="FTDI conf file does not exist. Creating it."
        echo $MSG

        echo "# This file allows the ftdi_sio driver to use FT X series." > $FTDI_CONF_FILE
        echo $FTDI_CONF_OPTION >> $FTDI_CONF_FILE
        echo $FTDI_CONF_ALIAS >> $FTDI_CONF_FILE

        #reload the module for the changes to have effect
        modprobe -r ftdi_sio
        modprobe ftdi_sio
fi
```

If all that went well, then you can run in one terminal (it does not print anything):

`rosrun robotiq_c_model_control CModelRtuNode.py /dev/ttyUSB0`

And in another terminal:

`rosrun robotiq_c_model_control CModelSimpleController.py`

to get a simple menu to activate and move the gripper using the USB adaptor... the rest is up to you.

Cheers :beers:

james2254, 2016-08-01:
@carlosjoserg Thank you very much! I have not sloved it and I will try it recently.

wuguangbin1230, 2017-11-02:
＠james2254 How did you solve the problems mentioned above?
I also could not the robotiq gripper by USB

wuguangbin1230, 2017-11-02:
@james2254 How did you solve the problems mentioned above?
I also could not the robotiq gripper by USB

newuhe, 2018-01-10:
@carlosjoserg I followed your instructions mentioned above, but I still receives 
`Unable to connect to /dev/ttyUSB0`
though I can get `Bus 001 Device 024: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)
`

carlosjoserg, 2018-01-10:
@newuhe 

* Check that the gripper connected to the port that you use when calling 
`rosrun robotiq_c_model_control CModelRtuNode.py /dev/ttyUSB0`

* I have tested with the 2F-85mm, but I've also receive positive feedback with the 2F-140mm, both _new models_. Don't know about the old molde, and I don't think it'd work with the 3-fingers model.

newuhe, 2018-03-20:
I've sloved the problem, which arise from my socket version.

yamunamaccarana, 2018-03-20:
@newuhe I think I have exactly the same problems that you had. How did you manage to solve them?
Thanks

yamunamaccarana, 2018-03-21:
@newuhe [here](https://github.com/ros-industrial/robotiq/issues/121) you say that you found other useful packages. Can you please specify which package you used?