ros-industrial-attic/robotiq #49: Assimp reports no scene in robotiq_c2_model_visualization

davetcoleman, 2016-02-09:
When using Rviz, the meshes in robotiq_c2_model_visualization seem to be improperly exported from SolidWorkds/MeshLab or wherever they came from:

```
ros.rosconsole_bridge.console_bridge: Assimp reports no scene in package://robotiq_c2_model_visualization/meshes/robotiq_c2/collision/robotiq_85_gripper_adapter_plate.dae
ros.rosconsole_bridge.console_bridge: Assimp reports no scene in package://robotiq_c2_model_visualization/meshes/robotiq_c2/visual/robotiq_85_gripper_adapter_plate.dae
ros.rosconsole_bridge.console_bridge: Assimp reports no scene in package://robotiq_c2_model_visualization/meshes/robotiq_c2/collision/robotiq_85_gripper_joint_2_L.dae
ros.rosconsole_bridge.console_bridge: Assimp reports no scene in package://robotiq_c2_model_visualization/meshes/robotiq_c2/collision/robotiq_85_gripper_joint_3_L.dae
ros.rosconsole_bridge.console_bridge: Assimp reports no scene in package://robotiq_c2_model_visualization/meshes/robotiq_c2/collision/robotiq_85_gripper_joint_0_L.dae
ros.rosconsole_bridge.console_bridge: Assimp reports no scene in package://robotiq_c2_model_visualization/meshes/robotiq_c2/collision/robotiq_85_gripper_joint_1_L.dae
ros.rosconsole_bridge.console_bridge: Assimp reports no scene in package://robotiq_c2_model_visualization/meshes/robotiq_c2/collision/robotiq_85_gripper_joint_2_R.dae
ros.rosconsole_bridge.console_bridge: Assimp reports no scene in package://robotiq_c2_model_visualization/meshes/robotiq_c2/collision/robotiq_85_gripper_joint_3_R.dae
ros.rosconsole_bridge.console_bridge: Assimp reports no scene in package://robotiq_c2_model_visualization/meshes/robotiq_c2/collision/robotiq_85_gripper_joint_0_R.dae
ros.rosconsole_bridge.console_bridge: Assimp reports no scene in package://robotiq_c2_model_visualization/meshes/robotiq_c2/collision/robotiq_85_gripper_joint_1_R.dae
ros.rosconsole_bridge.console_bridge: Assimp reports no scene in package://robotiq_c2_model_visualization/meshes/robotiq_c2/collision/robotiq_85_gripper_adapter_plate.dae
ros.rosconsole_bridge.console_bridge: Assimp reports no scene in package://robotiq_c2_model_visualization/meshes/robotiq_c2/visual/robotiq_85_gripper_adapter_plate.dae
ros.rosconsole_bridge.console_bridge: Assimp reports no scene in package://robotiq_c2_model_visualization/meshes/robotiq_c2/collision/robotiq_85_gripper_joint_2_L.dae
ros.rosconsole_bridge.console_bridge: Assimp reports no scene in package://robotiq_c2_model_visualization/meshes/robotiq_c2/collision/robotiq_85_gripper_joint_3_L.dae
ros.rosconsole_bridge.console_bridge: Assimp reports no scene in package://robotiq_c2_model_visualization/meshes/robotiq_c2/collision/robotiq_85_gripper_joint_0_L.dae
ros.rosconsole_bridge.console_bridge: Assimp reports no scene in package://robotiq_c2_model_visualization/meshes/robotiq_c2/collision/robotiq_85_gripper_joint_1_L.dae
ros.rosconsole_bridge.console_bridge: Assimp reports no scene in package://robotiq_c2_model_visualization/meshes/robotiq_c2/collision/robotiq_85_gripper_joint_2_R.dae
ros.rosconsole_bridge.console_bridge: Assimp reports no scene in package://robotiq_c2_model_visualization/meshes/robotiq_c2/collision/robotiq_85_gripper_joint_3_R.dae
ros.rosconsole_bridge.console_bridge: Assimp reports no scene in package://robotiq_c2_model_visualization/meshes/robotiq_c2/collision/robotiq_85_gripper_joint_0_R.dae
ros.rosconsole_bridge.console_bridge: Assimp reports no scene in package://robotiq_c2_model_visualization/meshes/robotiq_c2/collision/robotiq_85_gripper_joint_1_R.dae
```

I believe @mcevoyandy knows the easy fix for this

shaun-edwards, 2016-02-09:
This sometimes happens when the file path is too long. You could try moving
your workspace.

davetcoleman, 2016-02-09:
Indeed, I renamed the package from ``robotiq_c2_model_visualization`` to ``robotiq_c2`` and the problem went away. Note I did *not* change any filenames or folder names. I think I have a better fix than changing my root file path...