ros-industrial-attic/robotiq #55: Run error when loading 2F gripper

james2254, 2016-03-29:
when I run "roslaunch robotiq_c_model_control c_model_ethercat.launch" in the terminal,there is a error. what's wrong with it?
[image]

gavanderhoorn, 2016-03-29:
Have you remembered to build your workspace first? And then source the appropriate `setup.bash`?

PS: would you please copy/paste console output like this in the future? Screenshots aren't searchable, and they also make it hard to reply to issues by mail.

james2254, 2016-03-29:
@gavanderhoorn Thank you help me to point out flaws.I have built my workspace,and run "source devel/setup.bash "in the workspace.

gavanderhoorn, 2016-03-29:
> I have built my workspace,and run "source devel/setup.bash "in the workspace.

and you still get the same error?.

james2254, 2016-03-29:
Yes ,my ROS versions is indigo. I have download the soem and install the robotiq-modbus-tcp,then load   the robotiq package from https://github.com/ros-industrial/robotiq .when I  run "roslaunch robotiq_c_model_control c_model_ethercat.launch" in the terminal, still get the same error.

gavanderhoorn, 2016-03-29:
So you basically do:

```
mkdir /path/to/your/catkin_ws/src
cd /path/to/your/catkin_ws/src

git clone https://github.com/ros-industrial/robotiq.git

cd /path/to/your/catkin_ws

source /opt/ros/indigo/setup.bash
rosdep install --from-paths src --ignore-src

catkin_make

source devel/setup.bash
```

and then finally

```
roslaunch robotiq_c_model_control c_model_ethercat.launch
```
in that same terminal? And that fails with the error message you showed?

[robotiq_c_model_control/CMakeLists.txt](https://github.com/ros-industrial/robotiq/blob/2586c4b4adba3da0ba1c99f5a334296449caaf32/robotiq_c_model_control/CMakeLists.txt#L64-L65) is missing `install(..)` statements, so running from an `install` space wouldn't work, but I'd expect a `devel` space to work normally.

After `source devel/setup.bash`, what is the output of `rospack find robotiq_c_model_control`?

james2254, 2016-03-30:
james@james-23-g131cn:~/catkin_ws$ rospack find robotiq_c_model_control
/home/james/catkin_ws/src/robotiq-indigo-devel/robotiq_c_model_control
This is the output.

gavanderhoorn, 2016-03-30:
Ok. And just to make sure: your workspace is basically setup as I described?

shaun-edwards, 2016-06-25:
Thanks @gavanderhoorn for helping out.  @james2254 is this still an issue (it's quite old).  Let me know, otherwise I will close this.

james2254, 2016-06-27:
I'm sorry for a long time not responding to you. @gavanderhoorn .It works well.You can close it. @shaun-edwards .Thank you very much.

closed 2016-06-27 by shaun-edwards