ros-industrial-attic/robotiq #68: catkin_make error

xsmxsm, 2016-06-02:
CMake Error at /opt/ros/indigo/share/soem/SoemTargets.cmake:34 (message):
  Some (but not all) targets in this export set were already defined.

  Targets Defined: simple_test;red_test;slaveinfo;eepromtool;ebox;firm_update

  Targets not yet defined: osal;oshw;soem

Call Stack (most recent call first):
  /opt/ros/indigo/share/soem/soem-config.cmake:12 (include)
  /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:75 (find_package)
  robotiq/robotiq_ethercat/CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/xsm/catkin_ws1/build/CMakeFiles/CMakeOutput.log".
See also "/home/xsm/catkin_ws1/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

gavanderhoorn, 2016-06-02:
```
CMake Error at /opt/ros/indigo/share/soem/SoemTargets.cmake:
```

It would seem this is an error in `soem`, and not in this (`robotiq`) package.

Also: when reporting problems with build infrastructure, it's a good idea to include information on your OS, its version, how you installed it, version of ROS, how you installed that, etc, etc.

xsmxsm, 2016-06-02:
but before I put the robotiq package into workspace , catkin_make successed , when I put the robotiq package into workspace , it show error.Why?

gavanderhoorn, 2016-06-02:
Because nothing in your workspace was using the `soem` package?

xsmxsm, 2016-06-02:
nothing

gavanderhoorn, 2016-06-02:
No I meant: you did not get an error before, because no other package uses the `soem` package (which is the one with the error).

gavanderhoorn, 2016-06-02:
Which OS are you trying to build this on? Does it have CMake version > 3?

xsmxsm, 2016-06-02:
ubuntu14.10  cmake version 2.8.12.2

shaun-edwards, 2016-12-22:
Is this still an issue?  Can this be closed?

piraka9011, 2017-01-10:
I'm not sure if this is the same error but it does involve `soem`

```
CMake Warning at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "soem" with any of
  the following names:

    soemConfig.cmake
    soem-config.cmake

  Add the installation prefix of "soem" to CMAKE_PREFIX_PATH or set
  "soem_DIR" to a directory containing one of the above files.  If "soem"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  robotiq/robotiq_ethercat/CMakeLists.txt:4 (find_package)

-- Could not find the required component 'soem'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "soem" with any of
  the following names:

    soemConfig.cmake
    soem-config.cmake

  Add the installation prefix of "soem" to CMAKE_PREFIX_PATH or set
  "soem_DIR" to a directory containing one of the above files.  If "soem"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  robotiq/robotiq_ethercat/CMakeLists.txt:4 (find_package)

```

shaun-edwards, 2017-01-16:
Closing this issue...if this is still an issue, please feel free to comment with more information and we will reopen it.

closed 2017-01-16 by shaun-edwards