ros-industrial-attic/robotiq #69: Displaying colors in Gazebo

TheDash, 2016-06-02:
Right now, the arm spawns with white instead of black. Do these files exist anywhere to change it to blakc/use coloring while having the articulated gripper?

shaun-edwards, 2016-06-02:
I'm going to need a little context.  What do you mean by arm?  What gripper are you using, S or C model?  By articulated, do you mean the model with the fingers that "move"?

TheDash, 2016-06-02:
My apologies, I meant gripper. I'm using the S model and would like it to be detailed or have the dae's like the real life model. Right now it spawns as white. It is for the model with fingers that move

shaun-edwards, 2016-06-02:
I believe (though I'm not sure), the models can be found here `robotiq/robotiq_s_model_visualization/meshes/s-model_articulated/`.  The models are STLs, which may explain why they aren't detailed/textured.  Meshlab can be used to create DAE files from STLs.  The result isn't pretty.

TheDash, 2016-06-03:
Ok, I'll take it as there are nothing (mesh wise) provided as yet and close this ticket.

closed 2016-06-03 by TheDash