ros-industrial-attic/robotiq #81: Error compiling from source, soem headers not found

carlosjoserg, 2016-07-27:
Hi, I'm on 16.04/Kinetic, that's why I had to compile from source.

Don't know how this worked before (I was on 14.04/Inidigo a week ago, and compiled from source robotiq, with soem from apt-get without problem), probably pointing to old commits or something. However, using the latest of both repositories, robotiq's and [soem](https://github.com/smits/soem) within the same catkin workspace didn't compile for me directly.

I checked the `${soem_INCLUDE_DIRS}` var, and it has already all paths to all sub-folders containing headers (namely soem, oshw, osal) actually set [here](https://github.com/smits/soem/blob/master/CMakeLists.txt#L7), so I don't think it is necessary to add the `soem/` [here](https://github.com/ros-industrial/robotiq/blob/jade-devel/robotiq_ethercat/CMakeLists.txt#L20), and the same for all of [these](https://github.com/ros-industrial/robotiq/blob/jade-devel/robotiq_ethercat/src/ethercat_manager.cpp#L10-L18). In fact, I deleted those and then it compiled everything.

I ran also `catkin_make install`, and I have all necessary headers under `catkin_ws/install/include/soem`, without sub-paths.

I can provide a PR for this, but I haven't tried functionality still. Let me know.

shaun-edwards, 2016-07-28:
@carlosjoserg, thanks for the heads up.  We are still in catch-up mode for jade and kinetic.  Could this be related/same issue as addressed in PR #77

carlosjoserg, 2016-07-28:
Hi @shaun-edwards , not really sure. The description looks like, but certainly that PR is not addressing this issue for sure, and neither the file attached by the sender.

ipa-mdl, 2016-12-29:
I agree with @carlosjoserg, using `${soem_INCLUDE_DIRS}` should be correct.
`${soem_INCLUDE_DIRS}/soem` appends a string to a list(?).

gavanderhoorn, 2016-12-29:
Nice catch. This would probably completely resolve the build issues in #88 (and make ros-industrial/industrial_ci#87 a non-issue).

gavanderhoorn, 2016-12-29:
@shaun-edwards: I think this is something that would need to be fixed in any case, and it would also solve our other issue.

shaun-edwards, 2016-12-30:
@ipa-mdl, @gavanderhoorn, thanks for the bump.  I will work on this today.

shaun-edwards, 2016-12-30:
I made the changes described in this issue, and while they built just fine on my system, [all](https://travis-ci.org/ros-industrial/robotiq/builds/187744093) travis jobs failed.

To summarize:
- Previous version: 726cab91f38406906197af9f4bd25b1146837ec4 - CMake hacks with directory prefix on includes
    - `catkin_make` (all source) - fails (can't  find includes)
    - `catkin_make` (soem install - indigo only) - builds
    - `catkin build` (w/`catkin config --no-install`) - fails (can't find includes)
    - `catkin build` (w/`catkin config --install`) - builds
    - travis (fails - jade/kinetic versions)
 - Latest version: 64d5359892c66fc45787186c5dfa560a11edf352 - standard CMakeLists and no directory prefix on includes
    - `catkin_make` (all source) - builds
    - `catkin_make` (soem install - indigo only) - fails (can't  find includes)
    - `catkin build` (w/catkin config --no-install) - builds
    - `catkin build` (w/`catkin config --install`) - fails (can't  find includes)
    - travis (fails - all version)

 There's still an issue that is caused by the header directory structure differences between installed and source versions.  I'm still working on this, but I wanted to update my progress.

shaun-edwards, 2016-12-30:
I just noticed the specific include that causes the build to fail is different based on the type of build.  Sometimes the build fails on an soem include from the robotiq header and other times it fails on a soem include from an internal (soem) header.

ipa-mdl, 2016-12-31:
@shaun-edwards: have you tried `find_package(soem REQUIRED)` instead of the soem component?
I am not sure if `${soem_INCLUDE_DIRS}` is set by catkin or via https://github.com/smits/soem/blob/master/soem-config.cmake.in
It would be interesting to see to content of `${soem_INCLUDE_DIRS}` and `${catkin_INCLUDE_DIRS}` for some of the different cases.

shaun-edwards, 2017-01-16:
I'm going to take a look at this again later in the week.  However, given the lack of release for later versions for ROS, we might consider switching to a more actively maintained version [here](https://github.com/OpenEtherCATsociety/SOEM).

Kukanani, 2017-02-06:
Lack of Robotiq support is a major reason our lab has not moved to ROS Kinetic.

I forked the SOEM repository Shaun mentioned and catkinized it (fairly straightforward). The robotiq package now builds with no issues, but I haven't run any tests yet. We've made a `kinetic-devel` branch on our [fork](https://github.com/UTNuclearRoboticsPublic/robotiq/tree/kinetic-devel) of this repo.

rarrais, 2017-02-16:
Hi @Kukanani ,

I'm trying to compile your fork on Ubuntu 16.04 using Kinetic and I'm having a little trouble. Which SOEM repository are you using? Can you share the catkinized version?

Thank you!

carlosjoserg, 2017-02-16:
Hi @rarrais , I have been compiling in 16.04/Kinetic succesfully using the change in the `CMakeLists.txt` I proposed above using https://github.com/smits/soemst the latest, which you can find here https://github.com/beta-robots/robotiq/commit/a1275867de877569e9a737cc2c072ba80b68c32d , though I haven't tried the SOEM interface yet, I don't think the change would be an issue in functionality (I have been using USB)

Kukanani, 2017-02-16:
I have a 404 error attempting to access https://github.com/smits/soemst.

As I mentioned above, I catkinized the SOEM library at https://github.com/OpenEtherCATsociety/SOEM. I'll try to port into a public repo today.

I had to change the package name to lowercase, and adjusted the include directories of the robotiq package to match the new catkin package.

carlosjoserg, 2017-02-16:
Sorry, that's a typo, without the st: https://github.com/smits/soem

Kukanani, 2017-02-16:
Ah, thanks.

Here's our (now public) fork of OpenEtherCATsociety/SOEM: https://github.com/UTNuclearRoboticsPublic/soem. The commit with most of the catkinization is at https://github.com/UTNuclearRoboticsPublic/SOEM/commit/d3b67bd1dd0bf45ed3dc79c19ba7b4fdea949f7f. A Kinetic version of this (robotiq) repo is at https://github.com/UTNuclearRoboticsPublic/robotiq/tree/kinetic-devel, with the one extra commit pointing Robotiq to the new soem.

Kukanani, 2017-03-30:
I wanted to report that we've been using our catkinized version of OpenEtherCATsociety/SOEM on Robotiq S-Model hardware, with no issues so far.

felixvd, 2018-05-15:
@Kukanani Could you elaborate how you compiled on kinetic? I haven't been able to make it work.

closed 2018-05-15 by jproberge