ros-industrial-attic/robotiq #86: Issue with importing robotiq_cv_model_macro.xacro into my .xacro

atomoclast, 2016-11-03:
Hello all, I am trying to make a .xacro file that joins a UR10, a FT150 and a gripper. I am able to attatched and build the urdf as well as use check_urdf to verify the tree. The issue is the second I try to add in the gripper I keep getting this issue:

`Error:   Failed to find root link: Two root links found: [robotiq_85_adapter_link] and [world]
         at line 238 in /build/buildd/urdfdom-0.2.10+dfsg/urdf_parser/src/model.cpp
ERROR: Model Parsing the xml failed
`

I have no idea what's wrong with the way the file is defined to where it's considering it a world. It looks written properly.

gavanderhoorn, 2016-11-03:
Could you tell us a bit more (or better yet: show us) how you are combining the three xacro macros into one?

shaun-edwards, 2016-12-22:
Is this still an issue...can this be closed?

shaun-edwards, 2017-01-16:
Closing this issue...if this is still an issue, please feel free to comment with more information and we will reopen it.

closed 2017-01-16 by shaun-edwards