ros-industrial-attic/robotiq #90: English comments

NikolasE, 2017-01-04:
The comments are in english and french and I think they should be a) in a single language and b) in english.

shaun-edwards, 2017-01-05:
@NikolasE, I took a quick look and couldn't find any French comments.  Can you provide a link?  

Do you have a particular concern?  The original developer spoke French, so I assume that is why they are in there.

NikolasE, 2017-01-05:
https://github.com/ros-industrial/robotiq/blob/jade-devel/robotiq_force_torque_sensor/src/rq_sensor_com.cpp#L440

shaun-edwards, 2017-01-05:
From your original post, I thought there were the same comments in two different languages.  This would certainly be impossible to maintain for non-speakers.  I think converting french only comments to english is reasonable.  I normally wouldn't suggest we override the original author's intent, but since most comments are in english, I expect this was just an oversight.  

This could be done as needed (whenever someone is making changes) or on whole is someone wanted to take the time to google translate.  I'd accept a PR on this.

LazyEngineerToBe, 2019-12-12:
@shaun-edwards This issue could be closed.

gavanderhoorn, 2019-12-15:
@shaun-edwards is no longer maintaining this repository. That would be @jproberge.

closed 2019-12-15 by gavanderhoorn