ros-industrial-attic/robotiq #94: launch folder/file missing for robotiq_s_model_visualization in indigo-devel

gameofsong, 2017-01-13:
trying to roslaunch the 3 finger gripper in indigo-devel in the "robotiq_s_model_articulated_gazebo" folder and then it calls for a missing launch file "robotiq_gripper_upload.launch" in the "robotiq_s_model_visualization" folder.

tried to copy the one in the "groovy-devel" branch, which doesn't work for indigo-devel

can anyone help? 

`while processing /home/as/ros_industrial_ws/src/robotiq/robotiq_s_model_visualization/launch/robotiq_gripper_upload.launch:
Invalid roslaunch XML syntax: [Errno 2] No such file or directory: u'/home/as/ros_industrial_ws/src/robotiq/robotiq_s_model_visualization/launch/robotiq_gripper_upload.launch'`

shaun-edwards, 2017-01-13:
It's odd that the file went missing. Do you know why the groovy version doesn't work?

gameofsong, 2017-01-13:
sorry, first time running ros with robotiq 3 finger, do you know how to resolve this issue? thanks a lot

gameofsong, 2017-01-13:
tried to trace the spcific folder commit back to a year ago and more, and this file seems to be missing for a long time...

as for the goovy version of the "robotiq_gripper_upload.launch", tried and the following message showed up
`while processing /home/as/ros_industrial_ws/src/robotiq/robotiq_s_model_visualization/launch/robotiq_gripper_upload.launch:
Invalid roslaunch XML syntax: not well-formed (invalid token): line 45, column 90
The traceback for the exception was written to the log file`

then got lost in the line 45, column 90 of "robotiq_gripper_upload.launch" with the following code.
`    <meta name="request-id" content="AF20923B:2DAC3:1333F407:5878D48E" data-pjax-transient>`

am I examining the wrong roslaunch file, or what's wrong with it? thanks a lot

gavanderhoorn, 2017-01-13:
@gameofsong:

> then got lost in the line 45, column 90 of "robotiq_gripper_upload.launch" with the following code.
`    <meta name="request-id" content="AF20923B:2DAC3:1333F407:5878D48E" data-pjax-transient>`

how did you download / install the files? That is not typical contents of a launch file.

gameofsong, 2017-01-13:
@gavanderhoorn 
just copied from the folder in the groovy version to the indigo version and roslaunch ... 

from the message it seems it was only asking for the "robotiq_gripper_upload.launch" file in the "robotiq_s_model_visualization" folder ...

and the one in the groovy version is the only one i can find

gavanderhoorn, 2017-01-13:
But how did you get a copy of the files in the first place? Did you use `git clone ..`, did you use `apt-get` or some other way?

gameofsong, 2017-01-13:
@gavanderhoorn 

just copied from the groovy-devel's repository and paste into my local folder

gavanderhoorn, 2017-01-13:
> just copied from the groovy-devel's repository

which repository? Could you please be specific? A URL, a directory path?

Also: which exact steps did you take to get a copy of the "groovy-devel's" repository?

---

This does not directly address your original issue, but I'm pretty sure the launch file you have there is not as it should be, and I'm trying to figure out what caused this.

gameofsong, 2017-01-13:
@gavanderhoorn 

this very repository, under groovy branch, same folder

https://github.com/ros-industrial/robotiq/blob/groovy-devel/robotiq_s_model_visualization/launch/robotiq_gripper_upload.launch

gavanderhoorn, 2017-01-13:
And you opened [groovy-devel/robotiq_s_model_visualization/launch](https://github.com/ros-industrial/robotiq/tree/groovy-devel/robotiq_s_model_visualization/launch), then right-clicked on `robotiq_gripper_upload.launch` and then *Save Link As..*, correct?

If so, you saved an HTML page, not the content of the launch file. For that, please use the [raw link](https://github.com/ros-industrial/robotiq/raw/groovy-devel/robotiq_s_model_visualization/launch/robotiq_gripper_upload.launch) from [groovy-devel/robotiq_s_model_visualization/launch/robotiq_gripper_upload.launch](https://github.com/ros-industrial/robotiq/blob/groovy-devel/robotiq_s_model_visualization/launch/robotiq_gripper_upload.launch).

gameofsong, 2017-01-13:
my bad, seems like I saved the html instead of the actual code ... my bad

will try again and update

gavanderhoorn, 2017-01-13:
The original issue of the file missing after `groovy-devel` still remains though.

gavanderhoorn, 2017-01-13:
That particular launch file was removed in 5d3be12280db37b87d6d28622121fb6528701d55 by @TheDash, as part of a clean-up of the packages in this repository.

It's curious, as the file is still referenced in at least [robotiq_s_model_articulated_gazebo/launch/robotiq_gripper_empty_world.launch](https://github.com/ros-industrial/robotiq/blob/cc6e7949750732a30348d3dd2a0c350605eb9df9/robotiq_s_model_articulated_gazebo/launch/robotiq_gripper_empty_world.launch), which was added by @TheDash in cc6e7949750732a30348d3dd2a0c350605eb9df9.

gameofsong, 2017-01-13:
the code in the groovy version refers to a missing/replaced file of "robotiq_hand.urdf.xacro"

and in the indigo-devel, seems to have quite a few files to be choose from, and only the "s-model_articulated.xacro" (partial) works.

by partially I mean, seems like there's some problem with the base reference or what, but the gripper is just freely laying on the floor ... not sure if this is intensional or by mistake ...

I'm new to robotiq but this is kind of an issue and hopefully someone more experienced with this repository could help to resolve this issue with the 3 finger in indigo ...

shaun-edwards, 2017-01-14:
I should be able to get to this next week. Sorry it's not right. Thanks for reporting.

shaun-edwards, 2017-01-26:
@ancorasir, I've taken a look at the gazebo sim (once I applied your fix).  The gripper lies on the floor because the URDF doesn't have a fixed link to the `world`, as described [here](http://gazebosim.org/tutorials/?tut=ros_urdf#RigidlyFixingAModeltotheWorld).  Honestly, I'm not sure what the value of a gazebo sim is with just a single (un-attached) gripper.  Perhaps the package is just meant to test/demonstrate a physics sim of an articulated hand.

I'll submit a PR with your fix.  Please let me know if you have learned anything since your last comment.

ancorasir, 2017-01-27:
@shaun-edwards , thank you very much for the clarification. I can fully understand if this package was originally meant for gazebo testing purpose (just a way of showing that gazebo CAN simulate robotiq gripper). Current robotiq software are mainly for industrial application, but not on robot simulation. 

I thought such a powerful underactuated gripper should have been well developed for ROS/Gazebo simulation. I'm sure there's definitely a way to do it with currently packages, but unfortunately I'm not that skilled enough :( 

I thought there should have been a need for such an integrated simulation matching actual hardware setup that includes arm + gripper for pick+place. @TheDash developed something similar with the Grizzly_ur10, but it remains a challenge to make it work in a simpler way, especially on the Robotiq gripper side of grasping simulation. 

Thanks a lot for you guys' awesome work :)

closed 2019-04-30 by jproberge