ros-industrial-attic/robotiq #106: lose calibration each time the robot starts

Thompson104, 2017-08-24:
Hey all,

I have a robotiq FT300 and I'm using Ubuntu 14.04 with ROS Indigo. I used the calibration procedure in the manual, load the program in polyscope and calibrate the sensor. After the calibration, everything works find. But whenever I restart my computer or the robot, I would lose this calibration. Fz can be 10 or 5 or 3 with no force input.

I powered the FT300 from the UR control box, and connect is through the USB connector provided by Robotiq to my computer. I wonder if you guys can help me out.

Thank you in advance.

best,

shaun-edwards, 2017-08-24:
@nicolaslauzier

nicolaslauzier, 2017-08-25:
Hi,

@Thompson104 Please contact Robotiq support at support@robotiq.com for help with our products.

Best regards,

Thompson104, 2017-08-25:
@shaun-edwards @nicolaslauzier 
Thank you for your attention. I just sent the email and hopefully we can get this solved. My defense is only 1 month from now lol.

closed 2018-02-11 by jproberge