ros-industrial-attic/robotiq #108: robotiq_s_model_articulated_msgs vs robotiq_s_model_control.msgs

Shentheman, 2017-09-24:
What are the differences between robotiq_s_model_articulated_msgs vs robotiq_s_model_control.msgs?
Is the **articulated** one a different gripper hardware from the regular one?

I am trying to simulate the gripper in gazebo by sending robotiq_s_model_articulated_msgs to /SMRobotOutput. However, when dealing with real robot, I have to send robotiq_s_model_control.msgs to /SMRobotOutput. Is there a gazebo simulator for the non-articulated gripper?

shaun-edwards, 2017-09-25:
@Shentheman, thanks for asking.  Documentation is certainly lacking.  Unfortunately, I'm not familiar with using Robotiq with Gazebo, so I can't answer your question.  Hopefully another user will be able to answer this.

wxmerkt, 2017-12-01:
The messages have the same content and thus the same md5 hash. You should be able to use them interchangeably. 
I definitely agree that this collection of packages requites some TLC and happy to put some time in if we can then move it towards being released on the build farm/as debians - e.g. unify the S-Model messages for control and articulated as ``robotiq_s_model_msgs`` and release them (they are e.g. required on clients that do not require the drivers and the whole meta package to be built).

```
$ rosmsg md5 robotiq_s_model_articulated_msgs/SModelRobotInput
4d0701156e580a420c48833f57bc83f3
$ rosmsg md5 robotiq_s_model_control/SModel_robot_input
4d0701156e580a420c48833f57bc83f3
```

shaun-edwards, 2017-12-01:
@wxmerkt, thanks for helping answer.  I totally agree with the improvements that you suggest.  The lack of progress is due to a lack of time on my part.  I'm happy to accept PRs submitted by the community.

wxmerkt, 2018-04-29:
On another note, I just noticed that the folks at Clearpath noticed that a field in the SModelOutput was missing and added it (incompatible with this repository/upstream):

https://github.com/DualUR5Husky/robotiq/commit/73e5219cfab7d9d19f3c1a6b76548c1288ff9b76

christian-rauch, 2019-02-10:
I just encountered the same issue and mixed up the `robotiq_3f_gripper_articulated_msgs` and `robotiq_3f_gripper_control`.
While the "input" messages have indeed the same hash, the "output" messages differ:
```
$ rosmsg md5 robotiq_3f_gripper_articulated_msgs/Robotiq3FGripperRobotOutput 
d0c643ca7dd9fdb2d935922bf6659387
$ rosmsg md5 robotiq_3f_gripper_control/Robotiq3FGripper_robot_output
31ba91390a569c669af204c3d006a806
```
since the `rGLV` is missing in `robotiq_3f_gripper_articulated_msgs` as @wxmerkt pointed out.

I think it would make sense to go for a single set of messages for the very same purpose. I don't see the point in having twice the same messages. I am in favour of using the `robotiq_3f_gripper_articulated_msgs` since they already contain documentation about message content.

Would you accept a PR that replaces the `robotiq_3f_gripper_control` messages by `robotiq_3f_gripper_articulated_msgs` (including `rGLV`)?

PR #143 (2019-04-30): replace control msgs with robotiq_3f_gripper_articulated_msgs
There are two sets of input/output message definitions for the 3-finger gripper. This PR unifies these message definitions and removes redundancy (Fixes #108 ).

It adds `rGLV` (glove mode) to the articulated msgs to make them compatible (by hash) to the control msgs and consequently replaces all occurances of `robotiq_3f_gripper_control/Robotiq3FGripper_robot_{input,output}` with `robotiq_3f_gripper_articulated_msgs/Robotiq3FGripperRobot{Input,Output}`.

Tested with python nodes `Robotiq3FGripperTcpNode`, `Robotiq3FGripperSimpleCon [... cut]
closed 2019-04-30 by jproberge