ros-industrial-attic/robotiq #112: Control of a 3-finger adaptive gripper using Ethernet/IP protocol

aravindsiv, 2017-10-25:
I have a Robotiq 3-finger gripper configured with the Ethernet/IP protocol. I understand that in order to make use of this meta-package, [the Modbus TCP/IP protocol needs to be configured](http://wiki.ros.org/robotiq/Tutorials/Control%20of%20an%20S-Model%20Gripper%20using%20the%20Modbus%20TCP%20Protocol). 

Is there any way I can make use of this meta-package using the Ethernet/IP protocol, or is there any way I can reconfigure the gripper to use the Modbus TCP/IP protocol?

EtSamson, 2017-10-25:
@aravindsiv if you want to reconfigure your Gripper to Modbus TCP/IP contact the tech support at Robotiq and they can help you (will need to re-flash the Gripper). Email support@robotiq.com

aravindsiv, 2017-10-27:
Thanks @EtSamson . I contacted Robotiq, and they were able to help me out. If anybody else needs this functionality, feel free to ping me directly.

closed 2017-10-27 by aravindsiv