ros-industrial-attic/robotiq #113: Simultaneously using two FT-150s

rkeatin3, 2017-11-02:
I'm trying to use two FT-150s from one computer, but there are problems with the code that currently prevent me from doing so: 

1. There isn't a way to specify which device the node should connect to. It looks like code linked to within [Issue 107](https://github.com/ros-industrial/robotiq/issues/107) solves that problem

2. All of the functions and variables seem to be static... Has this been addressed before?

shaun-edwards, 2017-11-03:
>There isn't a way to specify which device the node should connect to. It looks like code linked to within Issue 107 solves that problem

If you could verify the fix, I'm happy to merge it.  I don't have access to hardware, let alone multiple, so it's impossible for me to test

> All of the functions and variables seem to be static... Has this been addressed before?

No...is it causing a problem?

rkeatin3, 2017-11-03:
I have verified that the code linked to in issue 107 does properly initialize the node with the device handle loaded into the param server. Perhaps the point made [here](https://github.com/ros-industrial/robotiq/issues/107#issuecomment-332756114) is a good one though: this change is breaking in that it will force users to specify the device.

It looks like I have to eat crow on my second point, however: I expected the use of global variables would cause problems when running two nodes (as they *seem* to be sharing a file descriptor, data buffers, etc.), but the data looks good from both of the devices I'm using.

shaun-edwards, 2017-11-05:
@rkeatin3, can you think of a work around for the single device?  Is there a way to make it backwards compatible?

rkeatin3, 2017-11-05:
Though I haven't tried, I imagine falling back to the default behavior if the rosparam hasn't been populated should work. The author of the fix certainly found that behavior contemptible, but I'm not sure what other options there are if the user isn't expected to know the path to the device file.

rkeatin3, 2017-11-06:
It worked as expected. Should I create a pull request?

rkeatin3, 2017-12-12:
This issue has been resolved with [Pull Request 114](https://github.com/ros-industrial/robotiq/pull/114).

closed 2017-12-12 by rkeatin3