ros-industrial-attic/robotiq #115: Missing ${prefix} in robotiq_arg2f_140_model_macro.xacro

focke-85, 2017-11-14:
Hello there

I believe a ${prefix} is missing in line 180 of the file robotiq/robotiq_arg2f_model_visualization/urdf/robotiq_arg2f_140_model_macro.xacro.

`<mimic joint="${prefix}finger_joint" multiplier="-1" offset="0" />`

Correct me if I am wrong ;)

gavanderhoorn, 2017-11-14:
Seems you're correct.

If you would submit a PR we can merge that.

gavanderhoorn, 2017-11-14:
Suggestion for future issues/comments: always try to include a link to the file + line, at the revision you found the issue in. That makes it much easier to report these kind of things, and much easier for us to find what you are referring to.

gavanderhoorn, 2017-11-14:
And it would seem line 174 is also missing the prefix:

https://github.com/ros-industrial/robotiq/blob/8d13a8b4545fb472420f370bd794fec79e4977b6/robotiq_arg2f_model_visualization/urdf/robotiq_arg2f_140_model_macro.xacro#L174

shaun-edwards, 2017-11-15:
Fixed in MR #117

closed 2017-11-15 by focke-85