ros-industrial-attic/robotiq #116: osal.h: No such file or directory

angadhn, 2017-11-14:
System config: 16.04/Kinetic

I am trying to install the robotiq package from source by cloning the kinetic-devel repository (available here:https://github.com/jhu-lcsr/robotiq) in my `catkin_ws/src` and then running `catkin_make`. However, I get the following message: `fatal error: osal.h: No such file or directory` as well as `Invoking "make -j2 -l2" failed`. I'd appreciate any help that I can get with this. Here is the full output of running `catkin_make`:
```
Base path: /home/sscrovers/experimentalCatkinWorkspaces
Source space: /home/sscrovers/experimentalCatkinWorkspaces/src
Build space: /home/sscrovers/experimentalCatkinWorkspaces/build
Devel space: /home/sscrovers/experimentalCatkinWorkspaces/devel
Install space: /home/sscrovers/experimentalCatkinWorkspaces/install
####
#### Running command: "cmake /home/sscrovers/experimentalCatkinWorkspaces/src -DCATKIN_DEVEL_PREFIX=/home/sscrovers/experimentalCatkinWorkspaces/devel -DCMAKE_INSTALL_PREFIX=/home/sscrovers/experimentalCatkinWorkspaces/install -G Unix Makefiles" in "/home/sscrovers/experimentalCatkinWorkspaces/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/sscrovers/experimentalCatkinWorkspaces/devel
-- Using CMAKE_PREFIX_PATH: /home/sscrovers/experimentalCatkinWorkspaces/devel;/opt/ros/kinetic
-- This workspace overlays: /home/sscrovers/experimentalCatkinWorkspaces/devel;/opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/sscrovers/experimentalCatkinWorkspaces/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.6
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 11 packages in topological order:
-- ~~  - robotiq (metapackage)
-- ~~  - robotiq_c2_model_visualization
-- ~~  - robotiq_ethercat
-- ~~  - robotiq_c_model_control
-- ~~  - robotiq_force_torque_sensor
-- ~~  - robotiq_modbus_rtu
-- ~~  - robotiq_modbus_tcp
-- ~~  - robotiq_s_model_control
-- ~~  - robotiq_joint_state_publisher
-- ~~  - robotiq_action_server
-- ~~  - robotiq_s_model_visualization
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'robotiq'
-- ==> add_subdirectory(robotiq/robotiq)
-- +++ processing catkin package: 'robotiq_c2_model_visualization'
-- ==> add_subdirectory(robotiq/robotiq_c2_model_visualization)
-- +++ processing catkin package: 'robotiq_ethercat'
-- ==> add_subdirectory(robotiq/robotiq_ethercat)
-- +++ processing catkin package: 'robotiq_c_model_control'
-- ==> add_subdirectory(robotiq/robotiq_c_model_control)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- robotiq_c_model_control: 2 messages, 0 services
-- +++ processing catkin package: 'robotiq_force_torque_sensor'
-- ==> add_subdirectory(robotiq/robotiq_force_torque_sensor)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- robotiq_force_torque_sensor: 1 messages, 1 services
-- +++ processing catkin package: 'robotiq_modbus_rtu'
-- ==> add_subdirectory(robotiq/robotiq_modbus_rtu)
-- +++ processing catkin package: 'robotiq_modbus_tcp'
-- ==> add_subdirectory(robotiq/robotiq_modbus_tcp)
-- +++ processing catkin package: 'robotiq_s_model_control'
-- ==> add_subdirectory(robotiq/robotiq_s_model_control)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- robotiq_s_model_control: 2 messages, 0 services
-- +++ processing catkin package: 'robotiq_joint_state_publisher'
-- ==> add_subdirectory(robotiq/robotiq_joint_state_publisher)
-- +++ processing catkin package: 'robotiq_action_server'
-- ==> add_subdirectory(robotiq/robotiq_action_server)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- +++ processing catkin package: 'robotiq_s_model_visualization'
-- ==> add_subdirectory(robotiq/robotiq_s_model_visualization)
-- Configuring done
CMake Warning (dev) at robotiq/robotiq_force_torque_sensor/CMakeLists.txt:54 (add_dependencies):
[... 73 lines cut ...]
[  1%] Built target std_msgs_generate_messages_nodejs
Scanning dependencies of target roscpp_generate_messages_lisp
[  1%] Built target roscpp_generate_messages_lisp
Scanning dependencies of target std_msgs_generate_messages_eus
[  1%] Built target std_msgs_generate_messages_eus
Scanning dependencies of target std_msgs_generate_messages_lisp
In file included from /home/sscrovers/experimentalCatkinWorkspaces/src/robotiq/robotiq_ethercat/src/ethercat_manager.cpp:10:0:
/opt/ros/kinetic/include/soem/ethercattype.h:68:18: fatal error: osal.h: No such file or directory
 #include <osal.h>
                  ^
compilation terminated.
[  1%] Built target std_msgs_generate_messages_lisp
robotiq/robotiq_ethercat/CMakeFiles/robotiq_ethercat.dir/build.make:62: recipe for target 'robotiq/robotiq_ethercat/CMakeFiles/robotiq_ethercat.dir/src/ethercat_manager.cpp.o' failed
make[2]: *** [robotiq/robotiq_ethercat/CMakeFiles/robotiq_ethercat.dir/src/ethercat_manager.cpp.o] Error 1
CMakeFiles/Makefile2:891: recipe for target 'robotiq/robotiq_ethercat/CMakeFiles/robotiq_ethercat.dir/all' failed
make[1]: *** [robotiq/robotiq_ethercat/CMakeFiles/robotiq_ethercat.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Scanning dependencies of target robotiq_force_torque_sensor_generate_messages_nodejs
[  3%] Generating Javascript code from robotiq_force_torque_sensor/sensor_accessor.srv
[  5%] Generating Javascript code from robotiq_force_torque_sensor/ft_sensor.msg
[  5%] Built target robotiq_force_torque_sensor_generate_messages_nodejs
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j2 -l2" failed
```

angadhn, 2017-11-14:
`$ ls /opt/ros/kinetic/include/soem -l' gives the following so I am not sure why osal.h is not found.
```
total 104
-rw-r--r-- 1 root root  5242 Jun 30 09:40 ethercatbase.h
-rw-r--r-- 1 root root  5521 Jun 30 09:40 ethercatcoe.h
-rw-r--r-- 1 root root  2784 Jun 30 09:40 ethercatconfig.h
-rw-r--r-- 1 root root  7526 Jun 30 09:40 ethercatconfiglist.h
-rw-r--r-- 1 root root  2635 Jun 30 09:40 ethercatdc.h
-rw-r--r-- 1 root root  2676 Jun 30 09:40 ethercatfoe.h
-rw-r--r-- 1 root root 18584 Jun 30 09:40 ethercatmain.h
-rw-r--r-- 1 root root  2376 Jun 30 09:40 ethercatprint.h
-rw-r--r-- 1 root root  5349 Jun 30 09:40 ethercatsoe.h
-rw-r--r-- 1 root root 17586 Jun 30 09:40 ethercattype.h
-rw-r--r-- 1 root root  4923 Jun 30 09:40 nicdrv.h
-rw-r--r-- 1 root root  2531 Jun 30 09:40 osal.h
-rw-r--r-- 1 root root  1580 Jun 30 09:40 oshw.h
```

angadhn, 2017-11-14:
I dug around a bit and made two changes which at least seems to have helped me resolve the problem though I have to play around with it some more. Here are the changes:
1. In '/opt/ros/kinetic/include/soem/ethercattype.h`, I changed `#include <osal.h>` to `#include </opt/ros/kinetic/include/soem/osal.h>`
2. In teh robotiq package that I got from the git repo in the first comment, I made the following change to the `(path_to_catkin_ws)/srcrobotiq/robotiq_ethercat/src/ethercat_manager.cpp` at line 11: change `include <oshw/linux/nicdrv.h>` to `#include <soem/nicdrv.h>`. I dunno if this is the right fix but it took away the errors. Any insights would be great though!

angadhn, 2017-11-14:
@shaun-edwards requesting some feedback because I see that you had encountered something like this on indigo based on https://github.com/smits/soem/issues/4

shaun-edwards, 2017-11-14:
I believe #81 addresses this in some detail.  If I remember correctly (it's been a while), I could not figure out how to get a pure source build (robotiq + soem) and a mixed build (robotiq + debian soem) to work.

angadhn, 2017-11-14:
thanks @shaun-edwards. I am looking into it. Do you have any experience working with the robotiq repo from the jhu-lcsr group and is that what you are using with kinetic? Also, any idea if there will be a kinetic branch at some point from ROS-I?

shaun-edwards, 2017-11-14:
I don't have any experience with the jhu-lcsr group.  It looks like they have a fork of the ROS-I repo with some extra commits.  I can create the kinetic branch, but I don't have a kinetic system to test.  Do you know if the jade_devel branch works under kinetic?

angadhn, 2017-11-14:
I am not sure I follow the bit about getting the source and mixed builds to work as I am still wet behind the years with ROS. However, I believe I have the debain soem based on robotiq's instructions to use apt-get and not the git repo of soem. In your experience, will the debain soem with jhu-lcsr robotiq be sufficient to get the gripper working via ROS?

shaun-edwards, 2017-11-14:
The source vs. debian building issue is caused by the location of header files.  It might be ROS build related, but the solution might be CMake trickery.  

I'm not familar with the jhu-lcsr so I can't say how well that will work.

angadhn, 2017-11-14:
Here is a snippet of the error that I get when I run catkin_make with the jade_devel clone in my catkin workspace. Not sure where to head with that.

```
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "socketcan_interface" with any of the following names:

    socketcan_interfaceConfig.cmake
    socketcan_interface-config.cmake
```

shaun-edwards, 2017-11-14:
It looks like a missing dependency, http://wiki.ros.org/socketcan_interface

Have you installed all the dependencies? Manually? or with rosdep: http://wiki.ros.org/rosdep#Install_dependency_of_all_packages_in_the_workspace

angadhn, 2017-11-14:
That fixed the dependency issue but still having some compiler issues as shown below:
```
[ 81%] Building CXX object robotiq/robotiq_s_model_articulated_gazebo_plugins/CMakeFiles/RobotiqHandPlugin.dir/src/RobotiqHandPlugin.cpp.o
In file included from /usr/include/c++/5/random:35:0,
                 from /usr/include/ignition/math2/ignition/math/Rand.hh:20,
                 from /usr/include/ignition/math2/ignition/math.hh:18,
                 from /usr/include/sdformat-4.0/sdf/Param.hh:34,
                 from /usr/include/sdformat-4.0/sdf/Element.hh:24,
                 from /usr/include/sdformat-4.0/sdf/sdf.hh:5,
                 from /usr/include/gazebo-7/gazebo/common/Plugin.hh:42,
                 from /home/sscrovers/jadeDevelRobotiqWS/src/robotiq/robotiq_s_model_articulated_gazebo_plugins/src/RobotiqHandPlugin.cpp:35:
/usr/include/c++/5/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
 #error This file requires compiler and library support \
```
This is followed by several more `error` and `note` messages.

angadhn, 2017-11-15:
I was able to resolve this compiler issue by adding the following line to the `CMakeLists.txt` in `robotiq/robotiq_s_model_articulated_gazebo_plugins/':
```
# Enable c++11 compiler
add_definitions(-std=c++11)
```

Also, I had another separate issue which was resolved by changing 'const' to 'constexpr' at lines [176](https://github.com/ros-industrial/robotiq/blob/f28e44bb637fbe7c259b36effa98ef046a390760/robotiq_s_model_articulated_gazebo_plugins/include/robotiq_s_model_articulated_gazebo_plugins/RobotiqHandPlugin.h#L176), [181](https://github.com/ros-industrial/robotiq/blob/f28e44bb637fbe7c259b36effa98ef046a390760/robotiq_s_model_articulated_gazebo_plugins/include/robotiq_s_model_articulated_gazebo_plugins/RobotiqHandPlugin.h#L181), [184](https://github.com/ros-industrial/robotiq/blob/f28e44bb637fbe7c259b36effa98ef046a390760/robotiq_s_model_articulated_gazebo_plugins/include/robotiq_s_model_articulated_gazebo_plugins/RobotiqHandPlugin.h#L184), and [187](https://github.com/ros-industrial/robotiq/blob/f28e44bb637fbe7c259b36effa98ef046a390760/robotiq_s_model_articulated_gazebo_plugins/include/robotiq_s_model_articulated_gazebo_plugins/RobotiqHandPlugin.h#L187).

angadhn, 2017-11-20:
@shaun-edwards fyi, I have added a kinetic-devel branch to[ my fork](https://github.com/angadhn/robotiq/tree/kinetic-devel) of the ROS-I robotiq repo and I have tested it on the gripper. I think people may also be interested in knowing that the Robotiq Wrist Camera is detected on Ubuntu, which I think is a good sign that it can be easily used via ROS for visual servoing. I'm looking more into that and can share updates if you think it is of interest to the community. At the same time, if you know of anyone who has already set up the camera, I would be very grateful for pointers.

closed 2017-11-21 by angadhn

qboticslabs, 2018-11-22:
Just change #include <osal.h>to

#include "osal.h". It will work