ros-industrial-attic/robotiq #118: Working fix: enables simultaneous operation of cam and gripper

angadhn, 2017-11-21:
Just wanted to offer a working fix (so far) for anyone with a similar issue/setup consisting of an arm with a robotiq wrist cam and gripper. I was having some issues where the gripper driver kept crashing while having the wrist camera stream running which was resolved by adding a couple lines as shown [here](https://github.com/angadhn/robotiq/commit/6b7508735909b91ce0bdd6482e4818a6ab29e0ee).

jproberge, 2018-04-15:
Thanks, this is indeed a simple and working patch that could be useful for some users!

closed 2018-04-15 by jproberge

angadhn, 2018-04-16:
@jproberge glad you found it useful.

I have another fix that someone might find useful for an alternate setup where the above fix didn't work.
I have a laptop which I use for all the algorithmic development in moveit, etc. The robot is hooked up to a separate PC  which is accessible over ssh (Since we share the robot in our lab, it keeps things nice and modular for people to test their work). For such a setup, the fixes in this link helped me have a mostly reliable operation of gripper and cam but it is not super robust. Usually restarting the drivers usually fixes it. Here is the fix: https://github.com/angadhn/starboy_hardware_ws/commit/8630cedadd7494968bfc5d73b1871e88fdf2272b

angadhn, 2018-04-16:
psuhed another bit of the fix that I'd forgotten to do a while back. https://github.com/angadhn/starboy_hardware_ws/commit/7d4fdf5007b1962f1b20118bd314afacfda2c4e9

iameijaz, 2023-12-28:
Am I too late to say that the links for those commits show a "404 page not found" error? I am working on the same setup except for the fact that I am using a ur5 which I can access over ssh as well. @angadhn can you kindly enlighten me how to access the ft-300 sensor values and the camera stream in rviz?