ros-industrial-attic/robotiq #120: ros_control implementation

philwall3, 2017-12-15:
Hi,
does anyone already have ros_control implemented for their real robotiq Gripper. I'm trying to do this now and I'm a bit lost on how to do it. 

Thanks!

jproberge, 2018-04-15:
Hello @philwall3 

I'm aware your message was written months ago, but in the meantime, Robotiq's maintainer of their ros packages has changed. So, do you still have any problem using a Robotiq gripper with ros?

philwall3, 2018-04-16:
Hi @jproberge  
I've managed to use the ros_control implementation from Carlos J. Rosales (https://github.com/beta-robots/robotiq).
This is my project: https://github.com/philwall3/UR5-with-Robotiq-Gripper-and-Kinect

philwall3, 2018-04-17:
1. I'm only using the ros_control implementation from Beta-Robots. It's instantiated here: https://github.com/philwall3/UR5-with-Robotiq-Gripper-and-Kinect/blob/master/myrobot2/myrobot2_driver/launch/myrobot2_bringup_ros_control.launch#L45

2. Since the ros_control implementation is done by @carlosjoserg and I'm just using it, I can't really help with that. Sorry.

closed 2018-07-20 by jproberge