ros-industrial-attic/robotiq #121: Unable to connect to /dev/ttyUSB0

newuhe, 2018-01-10:
I used `dmesg | grep tty` to check whether gripper is connected to my ubuntu pc and get
`[75163.432028] usb 1-3: FTDI USB Serial Device converter now attached to ttyUSB0`
however when I use `rosrun robotiq_c_model_control CModelRtuNode.py /dev/ttyUSB0` I still reeives 
`Unable to connect to /dev/ttyUSB0
/home/newuhe/catkin_ws/src/robotiq/robotiq_c_model_control/nodes/CModelRtuNode.py:65: SyntaxWarning: The publisher should be created with an explicit keyword argument 'queue_size'. Please see http://wiki.ros.org/rospy/Overview/Publishers%20and%20Subscribers for more information.
  pub = rospy.Publisher('CModelRobotInput', inputMsg.CModel_robot_input)
Traceback (most recent call last):
  File "/home/newuhe/catkin_ws/src/robotiq/robotiq_c_model_control/nodes/CModelRtuNode.py", line 89, in <module>
    mainLoop(sys.argv[1])
  File "/home/newuhe/catkin_ws/src/robotiq/robotiq_c_model_control/nodes/CModelRtuNode.py", line 75, in mainLoop
    status = gripper.getStatus()
  File "/home/newuhe/catkin_ws/src/robotiq/robotiq_c_model_control/src/robotiq_c_model_control/baseCModel.py", line 107, in getStatus
    status = self.client.getStatus(6);
  File "/home/newuhe/catkin_ws/src/robotiq/robotiq_modbus_rtu/src/robotiq_modbus_rtu/comModbusRtu.py", line 88, in getStatus
    response = self.client.read_holding_registers(0x07D0, numRegs, unit=0x0009)
  File "/usr/lib/python2.7/dist-packages/pymodbus/client/common.py", line 109, in read_holding_registers
    return self.execute(request)
  File "/usr/lib/python2.7/dist-packages/pymodbus/client/sync.py", line 82, in execute
    raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[rtu baud[115200]]
`

newuhe, 2018-01-11:
I used this instruction `sudo chmod 777 /dev/ttyUSB0` and retry, my gripper is connected but throws 
`AttributeError: 'NoneType' object has no attribute 'getRegister'`
and LED of the gripper is still red, which is blue when connected to my ur5 controller.
I checked it and found the issue happened in this line 
`response = self.client.read_holding_registers(0x07D0, numRegs, unit=0x0009)`
response gets NoneType and I don't know why, can anybody explain?

mch5048, 2018-01-31:
@newuhe I have same error with you. Did you solve it?

philwall3, 2018-02-19:
I had a similar problem when another launch-file was running which was accessing the USB-port. Try closing all other launch-files. Then start a fresh roscore and then `rosrun robotiq_c_model_control CModelRtuNode.py /dev/ttyUSB0`

angadhn, 2018-03-01:
I had similar issues when trying to activate the wristcam+gripper simultaneously; I was able to resolve it by throwing a small fix discussed [here](https://github.com/ros-industrial/robotiq/issues/118). It is likely not an elegant fix but just something that I was able to do to move on with using the gripper.

yamunamaccarana, 2018-03-20:
I have the same problems as newuhe.
I tried angadhn solution but now the execution of <code>robotiq_c_model_control CModelRtuNode.py</code> stops at the while cycle and the LED of the USBtoRS485 adapter stays red and never gets blue.

Any solutions?

newuhe, 2018-03-21:
I've solved the problem, it seems that it is because the socket package I used is different from the ros package [ros_canopen ](http://wiki.ros.org/ros_canopen)recommended by robotiq. Hope it will help you all.

closed 2018-03-21 by newuhe

yamunamaccarana, 2018-03-21:
@newuhe  can you please specify which package you used?

red0orange, 2024-04-25:
> @newuhe can you please specify which package you used?

`pip install pymodbus==2.2.0`

zhaianbang, 2024-05-29:
1-    sudo chmod 777 /dev/ttyUSB0
2-    pip install pymodbus==2.2.0

chenming0011112, 2024-10-18:
> 1- sudo chmod 777 /dev/ttyUSB0 2- pip install pymodbus==2.2.0 亲测可用, it worked!

您好！与上述同样的问题，ubuntu20.04的ros-noetic上按您的建议仍然报原来的错误：AttributeError: 'ModbusIOException' object has no attribute 'getRegister'。我能得到您的联系方式详细聊聊吗？

praj441, 2024-12-20:
> 1- sudo chmod 777 /dev/ttyUSB0 2- pip install pymodbus==2.2.0 亲测可用, it worked!

Not working

praj441, 2024-12-20:
To control robotiq gripper which is directly connected to the robot-wrist (UR), this worked for me:
https://dof.robotiq.com/discussion/2420/control-robotiq-gripper-mounted-on-ur-robot-via-socket-communication-python

Make sure to activate the gripper first from URCaps