ros-industrial-attic/robotiq #123: CMake Error while making robotiq_s_model_control because of socketcan_interface

yamunamaccarana, 2018-03-19:
I have ros kinetic installed on Ubuntu 16.04.3 LTS.
I installed:
$ easy_install -U pymodbus
and:
$ git clone https://github.com/ros-industrial/robotiq.git
in my catkin_ws/src.
However, when I catkin_make, I get the error down below*.
I also tried to:
$ git clone https://github.com/ros-industrial/ros_canopen.git
and catkin_make, but this bring me to other errors with variables set to NOTFOUND and include folders.

What should I do?

*Here the error I get;
**-- +++ processing catkin package: 'robotiq_s_model_control'
-- ==> add_subdirectory(robotiq/robotiq_s_model_control)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Error at /home/yami/catkin_ws/devel/share/socketcan_interface/cmake/socketcan_interfaceConfig.cmake:113 (message):
  Project 'socketcan_interface' specifies
  '/home/yami/catkin_ws/src/ros_canopen/socketcan_interface/include' as an
  include dir, which is not found.  It does neither exist as an absolute
  directory nor in
  '/home/yami/catkin_ws/src/ros_canopen/socketcan_interface//home/yami/catkin_ws/src/ros_canopen/socketcan_interface/include'.
  Check the issue tracker
  'https://github.com/ros-industrial/ros_canopen/issues' and consider
  creating a ticket if the problem has not been reported yet.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  robotiq/robotiq_s_model_control/CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/yami/catkin_ws_man/build/CMakeFiles/CMakeOutput.log".
See also "/home/yami/catkin_ws_man/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed**

ipa-mdl, 2018-03-19:
This is a [cross post](https://github.com/ros-industrial/ros_canopen/issues/271).
I was not able to reproduce this error.

jproberge, 2018-03-19:
Hello @yamunamaccarana,

This is indeed a cross post. Also, the errors mentioned in your post originate from the socketcan_interface package rather than the robotiq package. I understand that you've tried to install socketcan_interface in an attempt to fix the issues you were having during the compilation of the robotiq packages (if I'm not mistaken).

However, could you please first specify/reproduce the problems you were facing while compiling your catkin_ws that are specific to the robotiq package only?  If you have any troubles that are specific to robotiq, I might be able to better help you in that case.

yamunamaccarana, 2018-03-20:
Thankyou,

The error "CMake Error at /home/yami/catkin_ws/devel/share/socketcan_interface/cmake/socketcan_interfaceConfig.cmake:113 " that I posted at first happened before my attempt to install ros_canopen, while I was trying to compile the robotiq package only.

I will post the error after installing ros_canopen [here](https://github.com/ros-industrial/ros_canopen/issues/271)

yamunamaccarana, 2018-03-20:
ipa-mdl solved this [here](https://github.com/ros-industrial/ros_canopen/issues/271)

thanks

closed 2018-03-20 by yamunamaccarana