ros-industrial-attic/robotiq #125: More than 80% of the sampled goal states fail to satisfy the constraints imposed on the goal sampler. Is the constrained sampler working correctly?

AlexWang1996, 2018-04-14:
Hi all, l am just working on the robotiq-140 gripper with UR10 robot arm under ROS Kinectic.
[image]
l use moveit_setup_assistant to config the package. And run the demo.launch, l would like to see if l can plan the gripper in Moveit. l update the gripper to random goal state and click the plan button , and some errors happen like the following. 
[image]
After that l write some code to plan the gripper, and error happens,
[image]
[image]
so l am confused about it and really do not know how to deal with the problem.
can anyone helps me? Thank you.

AlexWang1996, 2018-04-16:
Actually, l try a lot of methods to solve this problem.
First, about the Constraint violated: l change the plan group(gripper) to only contain the finger joint, then l could plan the gripper in Moveit Rviz.
Second l was told that the plan cannot simply be executed because of lacking controllers. See [here ](https://github.com/beta-robots/robotiq/issues/7)

closed 2018-04-16 by AlexWang1996

jatinagarwal1308, 2020-12-10:
I am using ur5 2 finger gripper controller and getting the same error in python 
@bringBackm can you please guide me how to get rid of it?
Thanks in advance

bringBackm, 2020-12-14:
@jatinagarwal1308  
It seems to the wrong configuration.

shipeiZ, 2021-09-12:
Limits are wrong. Change lower limits opposite to upper limits for all joints except finger joint
take  inner_knuckle_joint as example
origin :<limit lower="0" upper="0.8757" velocity="2.0" effort="1000" />
modified :<limit lower="-0.8757" upper="0.8757" velocity="2.0" effort="1000" />