ros-industrial-attic/robotiq #132: FT300 never connects on ROS Kinetic/Ubuntu 16.04 using robotiq/jade-devel or kinetic-devel

otolithic, 2018-09-27:
I posted this on ROS answers and was told to bring it here. 

Our lab is using Kinetic on Ubuntu 16.04 and we can't connect to our FT300 sensor with this package. Using just the instructions in the FT300 manual (under the Development Package section), I was able to get sensor readings to show up in the terminal, so I do at least know my computer can make the connection. But `rosrun robotiq_force_torque_sensor rq_sensor`, which should do the same thing, just gives me lines and lines of "Waiting for sensor connection..." indefinitely. I've tried the following in various combinations:

- Using both the jade-devel and kinetic-devel branches of the Robotiq repository
- Specifying the exact \dev\ttyUSB* device that the sensor shows up as
- Modifying my files using TheDash's pull request that allows FT300 support

Unfortunately I lack the technical knowledge to know what the source of this problem is. As far as I can tell, the only fix is to downgrade to Ubuntu 14.04 and Indigo, since another lab has been able to get their FT300 working that way. Has anyone else run into this problem and managed to solve it somehow? Are there any tests I can run to figure out if ros-industrial/robotiq is actually the problem here? Otherwise, can anyone give me a rough idea of how much headway has been made towards making this possible?

Thanks for your attention.

otolithic, 2018-09-28:
Update: it appears the problem was one of permissions. Adding myself to the dialout was not enough to allow me to connect; I had to run `sudo chown [me] /dev/ttyUSB1`. I would close this issue except I don't know if it's one that can be fixed within the package itself or if this is an expected step in this type of connection.

gavanderhoorn, 2018-09-29:
> Update: it appears the problem was one of permissions. Adding myself to the dialout was not enough to allow me to connect; I had to run `sudo chown [me] /dev/ttyUSB1`. I would close this issue except I don't know if it's one that can be fixed within the package itself or if this is an expected step in this type of connection.
> 
> I think at the very least a line should be added to the wiki page for installation warning users of this.

I believe the reason this is not documented somewhere is that adding the user to the `dialout` group should be sufficient.

Perhaps @jproberge can comment on how he sets up his machines.

jproberge, 2018-10-01:
Hello @otolithic and @gavanderhoorn ,

First of all, I'm sorry for the delay, I was out of office and didn't see your post until now.

I frequently get emails from Robotiq customers with the exact same problem. Hence, this is indeed an issue many people encounter and I'll make sure to document it better in the wiki page. I'll start to significantly update the wiki page starting this week or next and I'll make sure to properly address this point.

So, as @gavanderhoorn already pointed out, this is solved by adding the $USER to the dialout group. This can be achieved normally simply by typing "**sudo usermod -a -G dialout $USER**". This is what I've done on my machine and this fix prevents you from having to give back read and write permissions each single time the device is plugged back in. I suggest you eventually use that solution over using commands of the kind "sudo chmod 666 /dev/ttyUSBX" because "X" will not be static and rather depends on the order in which ttyUSB devices are plugged in, and also, the permissions you give will be "forgotten" by the OS if you turn off your computer or even if you just unplug and re-plug in the FT300.

I'll make sure to add more details about that issue on the wiki page in the upcoming 5-7 days in an effort to minimize the number of people affected by this issue in the future. Thanks!

gavanderhoorn, 2018-10-01:
Thanks @jproberge.

Would it perhaps be an idea to add some `udev` rules to the repository as well?

Those could help with this issue.

jproberge, 2018-10-01:
@gavanderhoorn ,

Very good point and I totally agree with that idea! In fact, my initial answer to this issue was exactly to create proper udev rules. udev rules could fix non-persistent permissions and non-static device ID at the same time. I'll take the time to take a look at how this can be done properly within a ros package and I might indeed propose a pull request to add them. An eventual PR with valid udev rules could address this problem, but also indirectly address issues such as #107 .

jproberge, 2018-11-03:
I'm closing this issue, as suggested by @otolithic , more explicit information on how to resolve this issues has been added to the wiki page.

closed 2018-11-03 by jproberge

Gloriabhsfer, 2021-07-09:
Hi all:

I have plug the force sensor on my computer and follow the instruction of @jproberge by adding the user dialout and gave the permission of the USB port. Then I run the code:
`rosrun robotiq_ft_sensor rq_sensor`
It shows 
` [ INFO] [1625843512.095571884]: No device filename specified. Will attempt to discover Robotiq force torque sensor.
[ INFO] [1625843513.316304590]: Waiting for sensor connection...
`
My system is Kinetic (16.04), any ideas for getting connected to the force sensor?

Best,
Gloria