ros-industrial-attic/robotiq #136: Soem headers not working

atyshka, 2018-10-25:
I'm attempting to use this package in melodic and therefore had to compile soem from source. I compiled with cmake and everything seemed to go well. However, when I try to build this package I get 
`/home/alex/catkin_ws/src/robotiq/robotiq_ethercat/src/ethercat_manager.cpp:10:10: fatal error: soem/ethercattype.h: No such file or directory
 #include <soem/ethercattype.h>
`
So after a bit of digging I found [this](https://github.com/smits/soem/issues/8) where @shaun-edwards proposed a fix that was later added in a PR. However, after building from source on melodic I'm having this header issue again. Anyone else having this problem or have suggestions for a fix?

jproberge, 2018-10-25:
Hi @atyshka !

I unfortunately never had the chance to test our package stack on melodic yet. However, if you are building soem from source I might have a clue about what is happening here. 

First, could you just confirm that the "apparently-missing" header has really been installed on your machine? For example by simply typing 

`sudo updatedb; locate ethercattype.h`

Then, check this old PR #88 , especially the changes that was proposed to the file "robotiq_ethercat/CMakeLists.txt". If ethercattype.h can be found on your machine, I think I remember that when I used to compile soem from source, I had to remove the "/soem" part from "${soem_INCLUDE_DIRS}/soem". I'm a little bit guessing here since I don't have all the information, but your problem is an issue that I've seen before on other ros versions.

atyshka, 2018-10-25:
Sorry to waste your time, `make install` fixed it. I can't believe I forgot that.

closed 2018-10-25 by atyshka

felixvd, 2019-06-14:
@atyshka Could you be a bit more precise in what you did to fix this?