ros-industrial-attic/robotiq #139: Melodic support & gazebo 9.0 support

Jackey-Huo, 2018-12-22:
Hi, I use Ubuntu 18.04 and ros melodic with robotiq gripper. I soon found that, this official repo do not support melodic.

I try to build kinect-devel in my environment. There is something to be modified, but not too much. e.g. robotiq_s_model_articulated_gazebo_plugins still rely on gazebo::Math module, though it has been removed completely in gazebo 9.0...

I've successful compile it now. would you like a pull request about the melodic-devel?

christian-rauch, 2019-01-12:
I implemented melodic support in https://github.com/ros-industrial/robotiq/pull/133. This built successfully on my Ubuntu 18.04 / ROS melodic setup.
What compilation errors do you get?

Jackey-Huo, 2019-01-14:
@christian-rauch Oops, sorry for my stupid question. Your solution works perfectly well, my problem was solved. Close the issue

closed 2019-01-14 by Jackey-Huo