ros-industrial-attic/robotiq #147: Robotiq_3f_gripper shakes with the ur10 robot

padhupradheep, 2019-03-21:


jproberge, 2019-03-25:
Hi,

I have the exact same problem as the one you are describing with the 2f gripper. I'm still trying to find a solution about that, I'll write back as soon as I find one. Just to be clear though, is it the robot or the gripper that's vibrating in your case?

padhupradheep, 2019-03-25:
Hey,

Thanks for the response ! 

In my case, gripper is vibrating.

pidipidi, 2019-03-26:
I also faced the same issue when I use a UR5. There were two reasons. One is the position control of UR5/10 generates vibration in Gazebo. If you use joint effort control in simulation, you can suppress the physics issue. Another one is there can be self-collision between your gripper and UR10. If you disable it, you can remove the vibration. You can just disable it in ur.gazebo.xacro.

jproberge, 2019-04-30:
@pidipidi this indeed solved the issue on our side! Thanks for your comment!

closed 2019-04-30 by jproberge