ros-industrial-attic/robotiq #148: librobotiq_ethercat.so: undefined reference to `ec_group'

lazar505, 2019-03-23:
Hi,

I have a problem with installing robotiq library in ros melodic on Ubuntu 18.04. To be precise, when I run `catkin build robotiq`, it reports an error for robotiq_2f_gripper_control and robotiq_3f_gripper_control:

```
Errors     << robotiq_2f_gripper_control:make /home/lazar/catkin_ws/logs/robotiq_2f_gripper_control/build.make.028.log                                                
/home/lazar/catkin_ws/devel/.private/robotiq_ethercat/lib/librobotiq_ethercat.so: undefined reference to `ec_group'
/home/lazar/catkin_ws/devel/.private/robotiq_ethercat/lib/librobotiq_ethercat.so: undefined reference to `ec_config_map'
/home/lazar/catkin_ws/devel/.private/robotiq_ethercat/lib/librobotiq_ethercat.so: undefined reference to `ec_reconfig_slave'
/home/lazar/catkin_ws/devel/.private/robotiq_ethercat/lib/librobotiq_ethercat.so: undefined reference to `ec_init'
/home/lazar/catkin_ws/devel/.private/robotiq_ethercat/lib/librobotiq_ethercat.so: undefined reference to `ec_config_init'
/home/lazar/catkin_ws/devel/.private/robotiq_ethercat/lib/librobotiq_ethercat.so: undefined reference to `ec_readstate'
/home/lazar/catkin_ws/devel/.private/robotiq_ethercat/lib/librobotiq_ethercat.so: undefined reference to `ec_send_processdata'
/home/lazar/catkin_ws/devel/.private/robotiq_ethercat/lib/librobotiq_ethercat.so: undefined reference to `ec_configdc'
/home/lazar/catkin_ws/devel/.private/robotiq_ethercat/lib/librobotiq_ethercat.so: undefined reference to `ec_statecheck'
/home/lazar/catkin_ws/devel/.private/robotiq_ethercat/lib/librobotiq_ethercat.so: undefined reference to `ec_recover_slave'
/home/lazar/catkin_ws/devel/.private/robotiq_ethercat/lib/librobotiq_ethercat.so: undefined reference to `ec_receive_processdata'
/home/lazar/catkin_ws/devel/.private/robotiq_ethercat/lib/librobotiq_ethercat.so: undefined reference to `ec_slavecount'
/home/lazar/catkin_ws/devel/.private/robotiq_ethercat/lib/librobotiq_ethercat.so: undefined reference to `ec_slave'
/home/lazar/catkin_ws/devel/.private/robotiq_ethercat/lib/librobotiq_ethercat.so: undefined reference to `ec_close'
/home/lazar/catkin_ws/devel/.private/robotiq_ethercat/lib/librobotiq_ethercat.so: undefined reference to `ec_writestate'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/lazar/catkin_ws/devel/.private/robotiq_2f_gripper_control/lib/robotiq_2f_gripper_control/robotiq_2f_gripper_ethercat_node] Error 1
make[1]: *** [CMakeFiles/robotiq_2f_gripper_ethercat_node.dir/all] Error 2
make: *** [all] Error 2
cd /home/lazar/catkin_ws/build/robotiq_2f_gripper_control; catkin build --get-env robotiq_2f_gripper_control | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
......................................................................................................................................................................
Failed     << robotiq_2f_gripper_control:make                 [ Exited with code 2 ]                                                                                  
Failed    <<< robotiq_2f_gripper_control                      [ 2.0 seconds ]         
```                                                                           

I found out that `ec_...` comes from openssl so I installed it and added -lcrypto to compiler flag list and also edited cmake file to include openssl library, but the same error occurred. Additional information: I set in catkin config additional CMake args to: `-GEclipse CDT4 - Unix Makefiles -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_COMPILER_ARG1=-std=c++11 -D__cplusplus=201103L -D__GXX_EXPERIMENTAL_CXX0X__=1`

Can you help me with this issue or just tell me what can I also try here?

Thanks!

gavanderhoorn, 2019-03-24:
Just a clarification:

> I found out that `ec_...` comes from openssl [..]

No, not in this case. `ec` is short for "Ethercat".

jproberge, 2019-03-25:
Hi @lazar505 

I've never seen this error, but the problem seems to happen while compiling the specific package "robotiq_ethercat" and this package only has one dependency that could cause you trouble, which is soem. Thus, had you installed the soem dependency before building the workspace? 

You can install soem simply by typing "sudo apt-get install ros-melodic-soem". 

Please also check the other dependencies noted in the read.me. Hope this helps!

closed 2019-04-23 by jproberge