ros-industrial-attic/robotiq #150: robotiq_2f_140_gripper appears broken in Gazebo

minhkhang1795, 2019-04-19:
Hello, I'm trying to attach the **2f_140_gripper** to a UR5 arm. I got the URDF file of 2f_140_gripper from this repo and the UR5 from your `universal_robot` repo. 

Here is how I modified the [ur5_joint_limited_robot.urdf.xacro](https://github.com/olinrobotics/universal_robot/blob/hiro-xamyab/ur_description/urdf/ur5_joint_limited_robot.urdf.xacro) to attach the gripper to the UR5 arm's end effector:
```xml
<?xml version="1.0"?>
<robot xmlns:xacro="http://wiki.ros.org/xacro"
       name="ur5">
    ...
    <!-- robotiq 2f 140 gripper -->
    <xacro:include filename="$(find robotiq_2f_140_gripper_visualization)/urdf/robotiq_arg2f_140_model_macro.xacro"/>

    <link name="world"/>

    <joint name="world_joint" type="fixed">
        <parent link="world"/>
        <child link="base_link"/>
        <origin xyz="0.0 0.0 0.1" rpy="0.0 0.0 0.0"/>
    </joint>

    <xacro:robotiq_arg2f_140 prefix="gripper_"/>

    <joint name="gripper_ur5_joint" type="fixed">
        <parent link="tool0"/>
        <child link="gripper_robotiq_arg2f_base_link"/>
        <origin xyz="0 0 0.01" rpy="0 0 0"/>
    </joint>
</robot>
```
The result I got from this is: 
[image]

As you can see, some of the joints are broken. I did some research and found some people include [roboticsgroup_gazebo_plugins](https://github.com/waypointrobotics/robotiq_85_gripper/tree/master/robotiq_85_simulation/roboticsgroup_gazebo_plugins) _to add to Gazebo the mimic joint functionality that exists in URDF_. Basically, I cloned that repo and ran `catkin_make`. Then, I added the snippet below to `robotiq_arg2f_transmission.xacro`:
```xml
<?xml version="1.0"?>
<robot xmlns:xacro="http://ros.org/wiki/xacro">
  <xacro:macro name="robotiq_arg2f_transmission" params="prefix">
   ...

    <gazebo>
             <!-- loading plugin: mimic joints works in gazebo now-->
             <plugin filename="libroboticsgroup_gazebo_mimic_joint_plugin.so" name="${prefix}mimic_robotiq_140_4">
                <joint>${prefix}finger_joint</joint>
                <mimicJoint>${prefix}right_outer_knuckle_joint</mimicJoint>
                <!--<multiplier>-1.0</multiplier>-->
                <!--<offset>0.0</offset>-->
            </plugin>
            <plugin filename="libroboticsgroup_gazebo_mimic_joint_plugin.so" name="${prefix}mimic_robotiq_140_2">
                <joint>${prefix}finger_joint</joint>
                <mimicJoint>${prefix}left_inner_finger_joint</mimicJoint>
                <!--<multiplier>1.0</multiplier>-->
                <!--<offset>0.0</offset>-->
            </plugin>
            <plugin filename="libroboticsgroup_gazebo_mimic_joint_plugin.so" name="${prefix}mimic_robotiq_140_5">
                <joint>${prefix}finger_joint</joint>
                <mimicJoint>${prefix}right_inner_finger_joint</mimicJoint>
                <!--<multiplier>1.0</multiplier>-->
                <!--<offset>0.0</offset>-->
            </plugin>
            <plugin filename="libroboticsgroup_gazebo_mimic_joint_plugin.so" name="${prefix}mimic_robotiq_140_3">
                <joint>${prefix}finger_joint</joint>
                <mimicJoint>${prefix}left_inner_knuckle_joint</mimicJoint>
                <!--<multiplier>-1.0</multiplier>-->
                <!--<offset>0.0</offset>-->
            </plugin>
            <plugin filename="libroboticsgroup_gazebo_mimic_joint_plugin.so" name="${prefix}mimic_robotiq_140_5">
                <joint>${prefix}finger_joint</joint>
                <mimicJoint>${prefix}right_inner_knuckle_joint</mimicJoint>
                <!--<multiplier>-1.0</multiplier>-->
                <!--<offset>0.0</offset>-->
            </plugin>
        </gazebo>
  </xacro:macro>
</robot>
```
which results in the correct, connected gripper:
[image]

However, when I used MoveIt! to control the arm with the attached gripper shown above, MoveIt! fails to find the solution because the gripper collides with itself or something:

[image]

Note that if I remove the gripper, I can control the UR5 arm with MoveIt! normally. 

Do you have any idea why this doesn't work? If possible, could you please show me how to simulate the gripper using Gazebo and control it with MoveIt?

Thank you,
Khang Vu

--- 

Ubuntu: 16.04
ROS: Kinetic
Gazebo: 9.8.0 (the gripper is also broken in version 7)

minhkhang1795, 2019-04-20:
Hello,

I found out why the MoveIt! didn't work. I didn't disable collisions in the `ur5.srdf` file. For those who have similar problems with MoveIt!, you should add `<disable_collisions>` to your MoveIt! configuration. For example:

```xml
<disable_collisions link1="rgripper_right_outer_finger" link2="rgripper_robotiq_arg2f_base_link" reason="Adjacent"/>
<disable_collisions link1="rgripper_right_outer_knuckle" link2="rgripper_right_inner_knuckle" reason="Adjacent"/>
<disable_collisions link1="rgripper_right_outer_knuckle" link2="rgripper_right_outer_finger" reason="Adjacent"/>
<disable_collisions link1="rgripper_right_outer_finger" link2="rgripper_right_inner_finger" reason="Adjacent"/>
<disable_collisions link1="rgripper_right_inner_knuckle" link2="rgripper_right_inner_finger" reason="Adjacent"/>
<disable_collisions link1="rgripper_left_outer_knuckle" link2="rgripper_robotiq_arg2f_base_link" reason="Adjacent"/>
<disable_collisions link1="rgripper_left_inner_knuckle" link2="rgripper_robotiq_arg2f_base_link" reason="Adjacent"/>
<disable_collisions link1="rgripper_left_outer_knuckle" link2="rgripper_robotiq_arg2f_base_link" reason="Adjacent"/>
<disable_collisions link1="rgripper_left_outer_finger" link2="rgripper_robotiq_arg2f_base_link" reason="Adjacent"/>
<disable_collisions link1="rgripper_left_outer_knuckle" link2="rgripper_left_inner_knuckle" reason="Adjacent"/>
<disable_collisions link1="rgripper_left_outer_knuckle" link2="rgripper_left_outer_finger" reason="Adjacent"/>
<disable_collisions link1="rgripper_left_outer_finger" link2="rgripper_left_inner_finger" reason="Adjacent"/>
<disable_collisions link1="rgripper_left_inner_knuckle" link2="rgripper_left_inner_finger" reason="Adjacent"/>
```

gavanderhoorn, 2019-04-20:
> I found out why the MoveIt! didn't work. I didn't disable collisions in the `ur5.srdf` file. For those who have similar problems with MoveIt!, you should add `<disable_collisions>` to your MoveIt! configuration

whenever you change the urdf/xacro that a MoveIt configuration is based on, you should re-run the MoveIt Setup Assistant.

That would have taken care of this, as well as adding new groups and updating other settings that may be affected by the introduction of new links and joints.

Reusing a MoveIt configuration is almost never a good idea actually (and the ones in `universal_robot` are only ever meant as a quick test environment, not something that should be re-used).

closed 2019-04-20 by jproberge

yepw, 2019-07-08:
Hi, I faced the same problem and followed the solution above using `roboticsgroup_gazebo_plugins`. The gripper looks good in the static condition, but once I use the controller to close/open the gripper, it doesn't move properly. If you have the same problem, the following code might be helpful:

```ruby
<?xml version="1.0"?>
<robot xmlns:xacro="http://ros.org/wiki/xacro">
  <xacro:macro name="robotiq_arg2f_transmission" params="prefix">
    <transmission name="${prefix}finger_joint_trans">
      <type>transmission_interface/SimpleTransmission</type>
      <joint name="${prefix}finger_joint">
        <hardwareInterface>PositionJointInterface</hardwareInterface>
      </joint>
      <actuator name="${prefix}finger_joint_motor">
        <mechanicalReduction>1</mechanicalReduction>
      </actuator>
    </transmission>
	    <gazebo>
             <!-- loading plugin: mimic joints works in gazebo now-->
             <plugin filename="libroboticsgroup_gazebo_mimic_joint_plugin.so" name="${prefix}mimic_robotiq_140_4">
                <joint>${prefix}finger_joint</joint>
                <mimicJoint>${prefix}right_outer_knuckle_joint</mimicJoint>
                <multiplier>-1.0</multiplier>
            </plugin>
            <plugin filename="libroboticsgroup_gazebo_mimic_joint_plugin.so" name="${prefix}mimic_robotiq_140_2">
                <joint>${prefix}finger_joint</joint>
                <mimicJoint>${prefix}left_inner_finger_joint</mimicJoint>
                <multiplier>1.0</multiplier> 
            </plugin>
            <plugin filename="libroboticsgroup_gazebo_mimic_joint_plugin.so" name="${prefix}mimic_robotiq_140_1">
                <joint>${prefix}finger_joint</joint>
                <mimicJoint>${prefix}right_inner_finger_joint</mimicJoint>
                <multiplier>1.0</multiplier>
            </plugin>
            <plugin filename="libroboticsgroup_gazebo_mimic_joint_plugin.so" name="${prefix}mimic_robotiq_140_3">
                <joint>${prefix}finger_joint</joint>
                <mimicJoint>${prefix}left_inner_knuckle_joint</mimicJoint>
                <multiplier>-1.0</multiplier>
            </plugin>
            <plugin filename="libroboticsgroup_gazebo_mimic_joint_plugin.so" name="${prefix}mimic_robotiq_140_5">
                <joint>${prefix}finger_joint</joint>
                <mimicJoint>${prefix}right_inner_knuckle_joint</mimicJoint>
                <multiplier>-1.0</multiplier>
            </plugin>
        </gazebo>
  </xacro:macro>
</robot>
```

The complete package is [here](https://github.com/intuitivecomputing/ur5_with_robotiq_gripper). Hope this is helpful.

benediktkreis, 2022-05-10:
Is there a way to do the same with Ignition Gazebo Fortress? As far as I've seen the plugin libroboticsgroup_gazebo_mimic_joint_plugin.so is only available for Gazebo Classic.