ros-industrial-attic/robotiq #151: How to connect two robotiq FT sensors?

Chunting, 2019-04-20:
Hi, 
My working platform is:
Ubuntu: 16.04
ROS: Kinetic
URCap 3.8

I have two robotiq ft sensors, one FT 300 and one FT150.
Does this driver support to connect two sensors at the same time?
If so, how to seperate the data? Since it seems that the driver advertises the received date on the same topic "robotiq_ft_sensor" or "robotiq_ft_wrench" ?

Thanks,
Chunting

jproberge, 2019-04-20:
Hi @Chunting 

Unfortunately, the driver currently supports only a single force-torque sensor.

Now, to achieve what you want to do, you should probably look for information about how to run multiple instances of the same node. For example, see [this link](https://answers.ros.org/question/160071/running-multiple-instances-of-same-node/). As you will see, it's possible to run multiple instances as well as to remap each node's arguments such that, for example, the nodes and their messages won't have the same names.

gavanderhoorn, 2019-04-22:
@jproberge wrote:
> Unfortunately, the driver currently supports only a single force-torque sensor.

@jproberge: why do you say "unfortunately" here?

Running multiple instances of a single node to interface with multiple sensors seems like a proper way to implement a driver. And using namespaces and remapping are well-established practices in ROS, so unless a use-case makes it difficult, I personally always prefer to run multiple instances of a driver node that interfaces with a single piece of hw.

jproberge, 2019-04-22:
@gavanderhoorn , I said "unfortunately" as in "unfortunately [for you], the driver currently doesn't work this way [...]". I didn't imply that we should add a "multi-device functionality" to the driver or any other kind of modifications. I know running multiple instances is the proper way here, sorry if that wasn't clear.

This is a request I often get from customers and, in my experience, they are not all familiar nor comfortable with running multiple instances and remapping (otherwise, they probably wouldn't ask in the first place), even if it's well-established practices, which is why I said "unfortunately". Anyway, I don't think there should be any specific action taken.

Chunting, 2019-04-23:
Thanks for your reply.
I put the node in two different groups and now it works well.
```
	<group ns="$(arg right_controller_name)">
		<remap from="joint_states" to="/joint_states"/> 
		<node name="ur_hardware_interface" pkg="ur_modern_driver" type="ur_driver" output="log" launch-prefix="$(arg launch_prefix)">
			<param name="robot_ip_address" type="str" value="$(arg right_robot_ip)"/>
			<param name="reverse_port" type="int" value="$(arg right_reverse_port)" />
			<param name="min_payload" type="double" value="$(arg min_payload)"/>
			<param name="max_payload" type="double" value="$(arg max_payload)"/>
			<param name="max_velocity" type="double" value="$(arg max_velocity)"/>
			<param name="use_ros_control" type="bool" value="True"/>
			<param name="servoj_gain" type="double" value="750" />
			<param name="prefix" value="$(arg right_prefix)" />
			<param name="base_frame" type="str" value="$(arg right_base_frame)"/>
			<param name="tool_frame" type="str" value="$(arg right_tool_frame)"/>
			<param name="shutdown_on_disconnect" type="bool" value="$(arg shutdown_on_disconnect)"/>
		</node>
		<rosparam file="$(find dual_arm_robot_driver)/config/right_controllers.yaml" command="load"/>
		<node name="ros_control_controller_spawner" pkg="controller_manager" type="spawner" 
				respawn="false" output="screen" args="$(arg right_controllers)" />
		<node name="ros_control_controller_manager" pkg="controller_manager" type="controller_manager" respawn="false" output="screen" args="load $(arg right_stopped_controllers)" />
		<node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher"/>
		<!-- Start force/torque sensor --> 
		<node name="ft_sensor" pkg="robotiq_ft_sensor" type="rq_sensor" respawn="true" output="screen">
			<param name="frame_id" type="str" value="$(arg right_controller_name)_robotiq_ft_frame_id" />
			<param name="serial_id" type="str" value="ttyUSB0" />
		</node>
	</group> 
```

Thanks,
Chunting

jproberge, 2019-04-23:
Thanks for the update @Chunting !

I'm now closing this issue, good to know you were able to make it work :)

closed 2019-04-23 by jproberge