ros-industrial-attic/robotiq #154: ImportError: No module named robotiq_2f_gripper_control.baseRobotiq2FGripper

jokermay, 2019-05-07:
hello, I followed the tutorials "Control of a 2-Finger Gripper using the Modbus RTU protocol (ros kinetic and newer releases)" as [tutorials](http://wiki.ros.org/robotiq/Tutorials/Control%20of%20a%202-Finger%20Gripper%20using%20the%20Modbus%20RTU%20protocol%20%28ros%20kinetic%20and%20newer%20releases%29) , however, when i run the command "rosrun robotiq_2f_gripper_control Robotiq2FGripperRtuNode.py /dev/ttyUSB0" , it errors as 
`joker@joker:~/catkin_ws$ rosrun robotiq_2f_gripper_control Robotiq2FGripperRtuNode.py /dev/ttyUSB0
Traceback (most recent call last):
  File "/home/joker/catkin_ws/src/robotiq/robotiq_2f_gripper_control/nodes/Robotiq2FGripperRtuNode.py", line 50, in <module>
    import robotiq_2f_gripper_control.baseRobotiq2FGripper
ImportError: No module named robotiq_2f_gripper_control.baseRobotiq2FGripper
`
has anyone else met this problem? Why can not the node find the module "robotiq_2f_gripper_control.baseRobotiq2FGripper"?

Pranav186, 2020-07-15:
Hello, @jokermay , I am facing the same error. Did you find the solution to it?

ondrejba, 2023-01-22:
```
cd ~/catkin_ws
rosdep install --from-paths src --ignore-src -r -y
catkin_make
```