ros-industrial-attic/robotiq #159: Support for the Hand-E gripper

RaduCorcodel, 2019-07-08:
Is there any support for the Robotiq Hand-e gripper?

jproberge, 2019-09-15:
Hi @RaduCorcodel ,

Sorry for the late reply, but the package made for the 2f_gripper should work perfectly with the Hand-e gripper. I tried it a few months ago and it worked flawlessly. Tell me if you experience(d) any issues!

Best

closed 2019-09-15 by jproberge

RaduCorcodel, 2019-09-15:
Hi @jproberge. Thank you, yes the 2f gripper in the current development branch works with hand-e. We simply made some changes in the urdfs to load the correct geometry. It seems to be working perfectly. Since there is no option for hand-e in the main repo, do you want me to do a PR and integrate our implementation?

felipefelixarias, 2020-06-23:
Hi @RaduCorcodel, I am also looking for hand-e gripper support. Did you make your own urdf from the step files? If you could point me in the right directions I would be very grateful.

matthias-mayr, 2020-08-12:
@RaduCorcodel @felipefelixarias - Did you make some progress that you want to share?
At least the URDFs would be interesting - ideally also a simulation if anyone has worked on that.

@jproberge: I would not call this issue to be closed, unless robotiq is not willing to support the Hand-E gripper with ROS.
Even if it is compatible with the 2f_gripper package, this behavior should be documented and ideally descriptions being supplied.

felipefelixarias, 2020-08-12:
@matthias-mayr We were able to use this urdf (https://github.com/AcutronicRobotics/robotiq_modular_gripper) for our simulation. Unfortunately I cannot share our code right now. We had to change a few things since the urdf I mentioned uses ROS2. I am also not sure of how well this will translate to our physical deployment. It would indeed be nice to have a recommended/official approach for simulating and deploying the hande other than this issue @RaduCorcodel @jproberge

MOJOliciousFTW, 2020-08-24:
Hand-E works? I got the 2F 85 working, switched to Hand-E: no, not working. Guess I’ll try harder.

HeleneAlberdiArtola, 2021-03-27:
@felipefelixarias I am trying to control de hand-e robotiq gripper by using ros1 noetic and I was wondering if you managed to make the appropriate changes in (https://github.com/AcutronicRobotics/robotiq_modular_gripper) for its correct operation. If so I was wondering if you could provide me with the code. Thank you in advance.