ros-industrial-attic/robotiq #161: Control 2f gripper in RVIZ/Gazebo simulation

YueqiYu, 2019-07-24:
Hi, I'm able to install the robotiq 2f 140 gripper to ur5 model and do path planning or move ur5 robot in gazebo/ rviz, but don't know how to control this gripper to close and open, is there a specific code example? Thank you so much!

GDUTwd, 2020-03-26:
hi,I've been combining gripper with ur5 recently，but i get some trouble ,The gripper is shaking all the time，would you like to tell me the steps to  combine gripper and ur successfully? thank you so much!

ysl208, 2020-12-11:
Hi, i tried the robotiq_2f_gripper_action_server on my real UR5 robot, but it doesn't seem to work. 

It gives me a warning `gripper could not accept goal because the gripper is not yet active`, but I don't know how to activate it, it seems that I need to publish something in the /input topic

L-eonor, 2021-03-29:
You can control it by using a SimpleActionClient (http://wiki.ros.org/actionlib_tutorials). It works as following: http://wiki.ros.org/actionlib#Client-Server_Interaction

For that, you'll need to:
* add the  gripper controller to the .yaml file as following:
` # Gripper controller
gripper_controller:
  type: effort_controllers/GripperActionController
  gains:
    finger_joint: {p: 10,  d: 0.1, i: 1, i_clamp: 1}
  joint: finger_joint #important to call it joinT in the singular!
  action_monitor_rate: 20
  goal_tolerance: 0.002
  max_effort: 100
  stall_velocity_threshold: 0.001
  stall_timeout: 1.0
`
* the type must be a effort_controller or  a position_controller, according to the interface [interface](http://docs.ros.org/en/jade/api/hardware_interface/html/c++/classhardware__interface_1_1JointCommandInterface.html) specified in the transmission [file] (https://github.com/ros-industrial/robotiq/blob/66961ec6b6c9c493f13410d16ce32cea1648babf/robotiq_2f_85_gripper_visualization/urdf/robotiq_arg2f_transmission.xacro#L7)

* Send goal commands according to the message format [GripperCommandGoal](http://docs.ros.org/en/melodic/api/control_msgs/html/msg/GripperCommand.html)

Note: if grasping an object and commanding the gripper to fully close, the position won't be reached (gripper doesn't close fully because of the object). As consequence, won't be a result message returned from server.