ros-industrial-attic/robotiq #164: ROS support for robotiq epick vaccum gripper

gokul-gokz, 2019-09-03:
Are there ROS packages available for controlling robotiq epick vaccum gripper?

gokul-gokz, 2019-09-04:
I am planning to create a ROS package for the epick vacuum gripper and contribute to the ROS industrial community. So, before starting to work on writing the package, I just want to know , are there anyother people working on this issue, we can collaborate if needed

jproberge, 2019-09-07:
Hi, I'm sorry for not replying earlier, but I've already done one and I'm planning to do a PR soon about this. So, I developed them less than a month ago and it should be then considered as an "alpha release". As I haven't made a Pull Request to this repo they should not be considered official at this time, but they should work and I'd be glad to provide you assistance if you need it. You can find this new package, named "robotiq_vacuum_grippers" (as per Robotiq's request) on my personal Github for now : https://github.com/jproberge/robotiq  see the package robotiq_vacuum_grippers_control in this stack. 

I also recorded a short video in case you would like to check out the package as well as the EPick "in action": https://www.dropbox.com/s/8o2fn0ek5x9nobl/20190810_020640.mp4?dl=0 .

This package stack should work perfectly well on ROS Kinetic or ROS Melodic (I personally use Melodic). The procedure would be exactly the same as if you were to install our package stack from our official Github Repo. The only difference here is that you will have to use my own fork for now since, as I've said, it's not an official release yet. 

Finally here's a quick start guide to make your gripper work with ROS, which is also a big part of what I'm showing in the video:

1) Power up the vacuum gripper and connect it to your computer

2) Make sure you have the right permissions to read and write your EPick. As noted on the wiki page, this is needed to be done every time by typing, e.g.: "sudo chmod 666 /dev/ttyUSBO" without quotes, OR a single time if you just add your Linux user name to the "dialout" group by typing "sudo usermod -a -G dialout $USER". You just need to execute only one of the aforementioned commands depending on what you prefer the most (having to always give the permission when the gripper is connected or always having the proper rights by default).

3) Start ros by typing "roscore" in a terminal
4) Once your catkin workspace containing our ros package stack is compiled, you are now ready to run the sensor node. This is done by simply typing "rosrun robotiq_vacuum_grippers_control RobotiqVacuumGrippersRtuNode.py /dev/ttyUSB0"  without quotes (assuming ttyUSB0 is your gripper, which should be the case if you don't have another UART device plugged in). See the video if needed.

5) That's it! Now the gripper is listening to your commands! We provide a minimalist interface if you want to see some options for using your gripper (which I use in the video). You can run the interface by typing, in another terminal, "rosrun robotiq_vacuum_grippers_control RobotiqVacuumGrippersSimpleController.py". With this minimalist interface, you will be able to send and read commands to/from your gripper (video). You can also try "rosrun robotiq_vacuum_grippers_control RobotiqVacuumGrippersStatusListener.py" if you want to see that values of some of the input registers.

Best regards,

Jean-Philippe Roberge

gokul-gokz, 2019-09-12:
Hello,
Thanks for sharing the code. I was testing the code. I am getting the following error , incase if I am not sending any command for sometime,

> output.append((response.getRegister(i) & 0xFF00) >> 8)

> AttributeError: 'ModbusIOException' object has no attribute 'getRegister'

jproberge, 2019-09-15:
Thanks @gokul-gokz  for the additional information.

I've actually went to debug this problem at a customer's office this week and I've talked with some engineers from Robotiq about the issue as well. I will need to fix the problem, because there indeed seems to be a communication error that can happen once in a while. 

To temporarily solve the problem, I've made a patch for the customer which basically involved line 110 to 123 of the file baseRobotiqVacuumGrippers.py.  These are the problematic lines where a request will be sent to the vacuum gripper to send its status. Sometimes, it seems the gripper doesn't send back its status after it's been requested. So, a non-elegant but working patch for now (until a make a permanent fix) can simply be to modify those lines such that the process will continue to request the status from the gripper until the gripper sends back a response. This is clearly a patch, I will make sure to collaborate with Robotiq engineers to make a proper solution soon. This is just in the meantime, if you have to make this work.

I'll make a pull request this week and reference this issue. Also, you are right: after the gripper has been activated, you should first press "c" (Release) in order to grip again. This will be changed too. Thanks for your comments, you are helping making this package better.

li2jq, 2020-07-03:
Hi @jproberge 
Do you fix this problem for your customer?   _File "/home/li2jq/catkin_ws/src/robotiq-kinetic-devel/robotiq_modbus_rtu/src/robotiq_modbus_rtu/comModbusRtu.py", line 95, in getStatus
    output.append((response.getRegister(i) & 0xFF00) >> 8)
AttributeError: 'ModbusIOException' object has no attribute 'getRegister'
Also from your last comment, you made a patch to temporarily solve this problem. Could you provide the detailed instructions about this patch? Like where to add/change/delete the code. Thank you!

Best regards
Jiqing Li

robint-XNF, 2021-07-20:
> Hi, I'm sorry for not replying earlier, but I've already done one and I'm planning to do a PR soon about this. So, I developed them less than a month ago and it should be then considered as an [...]

hello，i have followed your steps, after i run command 'rosrun robotiq_vacuum_grippers_control RobotiqVacuumGrippersRtuNode.py /dev/ttyUSB0', the light turns to green, then ,i run "rosrun robotiq_vacuum_grippers_control RobotiqVacuumGrippersSimpleController.py", whatever command i use ,the epick didn't work. but i can use ur screen to control it. could you tell me why?

robint-XNF, 2021-07-21:
i also want to ask that why my epick's light turns to green rather than blue after i run 'rosrun robotiq_vacuum_grippers_control RobotiqVacuumGrippersRtuNode.py /dev/ttyUSB0', so i can't control my epick and also could not receive the status from the epick even though i have controlled it by ur.