ros-industrial-attic/robotiq #167: Can not connect gripper 2F-85 by USB

danfergo, 2019-10-13:
I recently got a 2F-85 Gripper, and I'm trying to get to work via USB/ROS.  My Issue Is similar to the ones described #45 and #121, but I followed all the steps and i can't seem to get the gripper to work. 

**I run:**
```
rosrun robotiq_2f_gripper_control Robotiq2FGripperRtuNode.py /dev/ttyUSB0
```
**Output:**
```
...:65: SyntaxWarning: The publisher should be created with an explicit keyword argument 'queue_size'. Please see http://wiki.ros.org/rospy/Overview/Publishers%20and%20Subscribers for more information.
  pub = rospy.Publisher('Robotiq2FGripperRobotInput', inputMsg.Robotiq2FGripper_robot_input)
Traceback (most recent call last):
  File "/home/.../robotiq/robotiq_2f_gripper_control/nodes/Robotiq2FGripperRtuNode.py", line 89, in <module>
    mainLoop(sys.argv[1])
  File "/home/.../robotiq/robotiq_2f_gripper_control/nodes/Robotiq2FGripperRtuNode.py", line 75, in mainLoop
    status = gripper.getStatus()
  File "/home/.../robotiq/robotiq_2f_gripper_control/src/robotiq_2f_gripper_control/baseRobotiq2FGripper.py", line 107, in getStatus
    status = self.client.getStatus(6);
  File "/home/.../robotiq/robotiq_modbus_rtu/src/robotiq_modbus_rtu/comModbusRtu.py", line 95, in getStatus
    output.append((response.getRegister(i) & 0xFF00) >> 8)
AttributeError: 'ModbusIOException' object has no attribute 'getRegister'

```
I've already added $USER to dialout and changed the  /dev/ttyUSB0 permitions to all sorts of configurations (currently I have it set to 777)
``` 
crwxrwxrwx 1 root dialout 188, 0 Oct 13 17:36 /dev/ttyUSB0
```

I'm running Ubuntu 18.04 and ROS Melodic (though I have experimented with non-ros scripts available on other the internet, and similar problems persist). I'm using the USB/Serial adapter that comes with the gripper. I've attempted to setup the modbus config file (as suggested here) https://github.com/ros-industrial/robotiq/issues/45#issuecomment-236564672 but the problem persists. The gripper LED stays Red the entire time. If I connect the gripper to the Universal Robots UR5, and control it using the URCaps, it works fine. 

**Connecting the USB and running `modprobe -r ftdi_sio`, `modprobe ftdi_sio` resultsin:**
```
[ 2257.006369] usb 9-2: new full-speed USB device number 6 using xhci_hcd
[ 2257.337193] usb 9-2: New USB device found, idVendor=0403, idProduct=6015
[ 2257.337194] usb 9-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2257.337195] usb 9-2: Product: USB TO RS-485
[ 2257.337195] usb 9-2: Manufacturer: FTDI
[ 2257.337195] usb 9-2: SerialNumber: DA3L6FJP
[ 2257.341461] ftdi_sio 9-2:1.0: FTDI USB Serial Device converter detected
[ 2257.341489] usb 9-2: Detected FT-X
[ 2257.341978] usb 9-2: FTDI USB Serial Device converter now attached to ttyUSB0
[ 2385.437036] usbserial: USB Serial deregistering driver FTDI USB Serial Device
[ 2385.437118] ftdi_sio ttyUSB0: FTDI USB Serial Device converter now disconnected from ttyUSB0
[ 2385.437129] usbcore: deregistering interface driver ftdi_sio
[ 2385.437164] ftdi_sio 9-2:1.0: device disconnected
[ 2385.463282] usbserial: USB Serial deregistering driver generic
[ 2385.463311] usbcore: deregistering interface driver usbserial_generic
[ 2426.515554] usbcore: registered new interface driver usbserial_generic
[ 2426.515561] usbserial: USB Serial support registered for generic
[ 2426.516663] ftdi_sio: unknown parameter 'product' ignored
[ 2426.516663] ftdi_sio: unknown parameter 'vendor' ignored
[ 2426.516700] usbcore: registered new interface driver ftdi_sio
[ 2426.516707] usbserial: USB Serial support registered for FTDI USB Serial Device
[ 2426.516744] ftdi_sio 9-2:1.0: FTDI USB Serial Device converter detected
[ 2426.516759] usb 9-2: Detected FT-X
[ 2426.517142] usb 9-2: FTDI USB Serial Device converter now attached to ttyUSB0
```
With the two warnings I'm under the impression that the modprobe config might not be being applied.

victor-robotic-24, 2019-10-25:
I am trying to control the Robotiq 2F-85 gripper in Ubuntu 16.04 (kinetic) and I also get the same error. I can't connect the gripper via USB. I can't put the gripper led in blue.
 
The first thing I do is install the following repository in my workspace: 

   `git clone https://github.com/ros-industrial/robotiq.git`

and the dependencies:

   `sudo apt-get install ros-kinetic-soem`
   `sudo easy_install -U pymodbus`

		robotiq_modbus_tcp (package included in the repository)

In order to connect the gripper to the USB serial port, read and write permissions are required to function properly. To do this, I add the username and execute the following command per terminal:

   `sudo usermod -a -G dialout $USER`
 
Next, to find out the port on which the controller is connected, I use: 

   `dmesg | grep tty`

Then I execute the following command to change the access permissions on the USB port:

   `sudo chmod 777 /dev/ttyUSB0`

Finally, after sending the usual roscore and source commands I launch this node to try to establish the connection:

   `rosrun robotiq_2f_gripper_control Robotiq2FGripperRtuNode.py /dev/ttyUSB0`

However, I get the following error and the gripper LED remains with the red light:

```
...65: SyntaxWarning: The publisher should be created with an explicit keyword argument 'queue_size'. Please see http://wiki.ros.org/rospy/Overview/Publishers%20and%20Subscribers for more information.
  pub = rospy.Publisher('Robotiq2FGripperRobotInput', inputMsg.Robotiq2FGripper_robot_input)
Traceback (most recent call last):
  File "/home/.../robotiq/robotiq_2f_gripper_control/nodes/Robotiq2FGripperRtuNode.py", line 89, in <module>
    mainLoop(sys.argv[1])
  File "/home/.../robotiq/robotiq_2f_gripper_control/nodes/Robotiq2FGripperRtuNode.py", line 75, in mainLoop
    status = gripper.getStatus()
  File "/home/.../robotiq/robotiq_2f_gripper_control/src/robotiq_2f_gripper_control/baseRobotiq2FGripper.py", line 107, in getStatus
    status = self.client.getStatus(6);
  File "/home/.../robotiq/robotiq_modbus_rtu/src/robotiq_modbus_rtu/comModbusRtu.py", line 95, in getStatus
    output.append((response.getRegister(i) & 0xFF00) >> 8)
AttributeError: 'ModbusIOException' object has no attribute 'getRegister'

```

danfergo, 2019-10-25:
Hi @victor-robotic-24,

I don't know what exactly did the trick, but I was able to get the gripper to work after connecting it to the UR5 and changing the device is in the gripper/urcaps settings. I also connected the gripper to the "Robotiq User Interface" on a Windows desktop, and made sure the protocol settings matched the ones on the manual description.

Best of luck.

Best, 
Daniel

tejaswid, 2020-03-13:
I occasionally have the same problem, even after granting correct permissions using the `chmod` and `usermod add` commands. The fix that worked for me was to physically disconnect the gripper and reconnect it. 

It seems that some internal flags are reset when the power to the gripper through the pins that connect it to the robot arm is turned off. (pins at the back of the gripper, not the cable)

Surprisingly, turning the robot arm off and turning it on did not have any affect. Not sure what exactly is going on. Just guessing that some internal state is being kept as long as the pins on the gripper are in contact with those of the arm.

gtatiya, 2020-03-18:
I was facing the exact same problem, and here's how I fixed it:

In line 65 of Robotiq2FGripperRtuNode.py (https://github.com/ros-industrial/robotiq/blob/66961ec6b6c9c493f13410d16ce32cea1648babf/robotiq_2f_gripper_control/nodes/Robotiq2FGripperRtuNode.py#L65), replace `pub = rospy.Publisher('Robotiq2FGripperRobotInput', inputMsg.Robotiq2FGripper_robot_input)` with `pub = rospy.Publisher('Robotiq2FGripperRobotInput', inputMsg.Robotiq2FGripper_robot_input, queue_size=10)`. 

Solution:
https://github.com/RethinkRobotics/baxter_examples/issues/32
https://github.com/RethinkRobotics/baxter_examples/pull/34

kamilcetin, 2020-09-01:
I had the same problem of 
AttributeError: 'ModbusIOException' object has no attribute 'getRegister'

I did all things above, but did not solve it. When I changed my RS485-USB interface board, it works now.

EmilyJrxx, 2020-10-20:
I ran into almost the same problem:

> rosrun robotiq_2f_gripper_control Robotiq2FGripperRtuNode.py /dev/ttyUSB0
Traceback (most recent call last):
  File "/home/xxwang/ROS_Workspaces/ros_k4a_auboi5_ws/src/robotiq/robotiq_2f_gripper_control/nodes/Robotiq2FGripperRtuNode.py", line 89, in <module>
    mainLoop(sys.argv[1])
  File "/home/xxwang/ROS_Workspaces/ros_k4a_auboi5_ws/src/robotiq/robotiq_2f_gripper_control/nodes/Robotiq2FGripperRtuNode.py", line 75, in mainLoop
    status = gripper.getStatus()
  File "/home/xxwang/ROS_Workspaces/ros_k4a_auboi5_ws/src/robotiq/robotiq_2f_gripper_control/src/robotiq_2f_gripper_control/baseRobotiq2FGripper.py", line 107, in getStatus
    status = self.client.getStatus(6);
  File "/home/xxwang/ROS_Workspaces/ros_k4a_auboi5_ws/src/robotiq/robotiq_modbus_rtu/src/robotiq_modbus_rtu/comModbusRtu.py", line 95, in getStatus
    output.append((response.getRegister(i) & 0xFF00) >> 8)
AttributeError: 'ModbusIOException' object has no attribute 'getRegister'

But my environment is a little bit different: My 2f-85 is not directly connected to my ROS Machine, but to my AUBO i5 controller, and I managed to send modbus commands via SOCAT following @jproberge 's instructions.  I tried all methods above, but did not resolve the issue. After reading this along with other posts, I'm guessing this 'getRegister' does not reveal the true issue, and we didn't ran into the same trouble actually. I found my true issue is that my gripper was not 'connected' to robot controller which should be done on the teach pendant. check [this](https://dof.robotiq.com/discussion/1914/robotiq-gripper-with-ur5-cb3-thru-wrist-connector?)   

Good luck to anyone who run into this 'getRegister' issue later, hope you can find your solution

ZZWang21, 2022-08-17:
> Hi @victor-robotic-24,
> 
> I don't know what exactly did the trick, but I was able to get the gripper to work after connecting it to the UR5 and changing the device is in the gripper/urcaps settings. I also connected the gripper to the "Robotiq User Interface" on a Windows desktop, and made sure the protocol settings matched the ones on the manual (RTU protocol).
> 
> Best of luck.
> 
> Best, Daniel

HI, @danfergo , sorry, i did not understand your reply. Did you mean that you connect the gripper to UR5, and connect the UR5 to your computer? How did you connect? And what is ROBOTIQ User Interface?  Thank you.

tammerb, 2022-10-28:
Hi all. This fork of the robotiq driver repo seems to be the most up to date.
https://github.com/TAMS-Group/robotiq

They appear to be doing their best to keep up to date with the pymodbus updates. All I had to do was comment out two lines:
`#        status = gripper.getStatus()`
`#        pub.publish(status)`

that are throwing attribute errors in [this file](https://github.com/TAMS-Group/robotiq/blob/noetic-devel/robotiq_2f_gripper_control/nodes/Robotiq2FGripperRtuNode.py). It'd be nice to get the status for but until they resolve some changes in pymodbus that's my workaround.

red0orange, 2024-04-25:
> Hi all. This fork of the robotiq driver repo seems to be the most up to date. https://github.com/TAMS-Group/robotiq
>
> They appear to be doing their best to keep up to date with the pymodbus updates. [...]

or `pip install pymodbus==2.2.0`