ros-industrial-attic/robotiq #173: Error when following Husky Dual UR5 Mobile Manipulation Demo (Simulation)

stevensu1838, 2019-12-18:
Hi, 
I've been following the Husky Dual UR5 Mobile Manipulation Demo (Simulation) tutorial. And I am on Controlling The Grippers session. I could control the gripper with no problem by running the following commands:
$ roslaunch husky_gazebo husky_empty_world.launch
$ rosrun robotiq_s_model_control SModelSimpleController.py _topic:=/left_hand/command
typing “a” will activate the gripper, and then “o” will open it. 

However, when I trying to load the robot in a more complicated environment instead of the empty world by the following: 
$ roslaunch husky_gazebo husky_playpen.launch
$ rosrun robotiq_s_model_control SModelSimpleController.py _topic:=/left_hand/command
When typing “a” to activate it, both of the two arms and gripers disappear in Gazebo. Any chance you know what's going on and how to fix it?

gavanderhoorn, 2019-12-18:
Cross-posted: [robot disappears in Gazebo when following Husky Dual UR5 Mobile Manipulation Demo (Simulation)](https://answers.ros.org/question/340239).

stevensu1838, 2019-12-20:
Hi Buddy, 
Is it because of the Christmas that no one answers this question?

jproberge, 2019-12-22:
Hi @stevensu1838 ,

I don't see how this could really be related to Robotiq's package stack, but I might be wrong. If you want more people to jump in and help, then could you provide more details? Which ROS/Gazebo versions are you running? Also, if you have followed the tutorial you pointed out to, you have installed a forked version of this repo (i.e.: https://github.com/DualUR5Husky/robotiq) which doesn't seem to have been updated for several years, and which I don't support either. I would recommend posting an issue on their specific repo in this case. What have you tried so far? Any errors / warnings apart from parts of your robot disappearing ?

GDUTwd, 2020-03-26:
hi,I've been combining gripper with ur5 recently，but i get some trouble ,The gripper is shaking all the time，would you like to tell me the steps to combine gripper and ur successfully? thank you so much!