ros-industrial-attic/robotiq #177: ImportError: No module named robotiq_2f_gripper_control.baseRobotiq2FGripper

Pranav186, 2020-07-15:
Hello,
 I am trying to control robotiq 2 finger gripper with ros kinetic. I followed the tutorial from roswiki. After running the command-
```
rosrun robotiq_2f_gripper_control Robotiq2FGripperRtuNode.py /dev/ttyUSB0
```
I am getting the following output on the terminal-
```
File "/home/pranav618/catkin_ws/src/robotiq/robotiq_2f_gripper_control/nodes/Robotiq2FGripperRtuNode.py", line 47, in <module>
    import robotiq_2f_gripper_control.baseRobotiq2FGripper
ImportError: No module named robotiq_2f_gripper_control.baseRobotiq2FGripper
```
Can someone help me with this ?

Pranav186, 2020-07-16:
Solved by creating a new workspace for the gripper.

closed 2020-07-16 by Pranav186

rolandying, 2021-05-04:
Hi, I have faced the same error, and I have tried to creat a new workspace, but it didn't make a difference, so could you tell me more details about how to solve the problem. Looking forward for your reply!