ros-industrial-attic/robotiq #185: Connect to 2f gripper via USB, using action server

jeff-o, 2021-02-23:
Hi! I would really appreciate some guidance on how to properly connect with a 2f gripper, so that it publishes joint state positions and can be controlled by publishing goal messages.

I know that the gripper is reachable, as I was able to connect with it by following this [tutorial](https://wiki.ros.org/robotiq/Tutorials/Control%20of%20a%202-Finger%20Gripper%20using%20the%20Modbus%20RTU%20protocol%20%28ros%20kinetic%20and%20newer%20releases%29). However, this only gets me halfway, as I need to control and visualize the gripper using moveit.

Do I need to launch the RTU node and action server separately? Or is there a launch file somewhere I haven't found that does it all? Thanks!

gavanderhoorn, 2021-02-24:
I'm not a maintainer here, but this repository is hosted on our organisation, so I'd like to do some expectation management.

Robotiq do not appear to support the packages in this repository any more. They haven't done so, for quite some time.

@jproberge took over for some time, but appears to also have switched to something else.

Other members of `ros-industrial` (ie: the Github organisation) are not in a capacity to assist unfortunately.

I would suggest to post your question on an appropriate Robotiq forum, or a community supported forum.

ZZWang21, 2022-08-17:
Hi, @jeff-o  have you got a solution? Thank you.

jeff-o, 2022-08-17:
Hopefully this helps!

Install the Clearpath Robotics' fork of the Robotiq driver. Note that this repository supports all of their
grippers, as well as the force-torque sensor.

Install necessary dependencies.
sudo apt-get install ros-melodic-gazebo-plugins
sudo apt-get install ros-melodic-socketcan-interface
sudo apt-get install ros-melodic-soem
pip install pymodbus

Create a workspace if you have not already.
mkdir -p ~/catkin_ws/src

Clone and build the driver in your workspace.
cd ~/catkin_ws/src
git clone https://github.com/clearpathrobotics/robotiq
cd ..
rosdep install --from-paths src --ignore-src --rosdistro=melodic -y -r
catkin_make
source devel/setup.bash

The Robotiq action server may be launched with the following. **Make sure that the launch file has been pre-configured to look at the correct device path.**

roslaunch robotiq_2f_gripper_control robotiq_action_server.launch

The gripper can be opened or closed by publishing to its action server:
rostopic pub /command_robotiq_action/goal robotiq_2f_gripper_msgs/CommandRobotiqGripperActionGoal <tab-complete>

Populate the header with the following parameters, then press <enter>:
emergency_release = False
stop = False
position = [0.00 to 0.085]
speed = [0.01 to 0.1]
force = 5.0