ros-industrial-attic/robotiq #189: Support for Noetic

RaduCorcodel, 2021-07-06:
Hello,

Are there any plans to support Ubuntu20/ROS-Noetic? Thank you.

Radu,

gavanderhoorn, 2021-07-06:
Please see the [Status](https://github.com/ros-industrial/robotiq/tree/45196f6558fe8ba9d89bc8a105396c68c3e7e892#status) section in the main readme.

I don't believe anything has changed.

RaduCorcodel, 2021-07-09:
Hi @gavanderhoorn,

I'm only asking because the newest branch which is kinetic-devel does not build on Noetic. A series of errors occur which start with the message bellow, then it's more of the same errors for pretty much the rest of the project.

```
In file included from /usr/include/ignition/math6/ignition/math/Quaternion.hh:20,
                 from /usr/include/ignition/math6/ignition/math/Pose3.hh:20,
                 from /usr/include/sdformat-9.5/sdf/Actor.hh:23,
                 from /usr/include/sdformat-9.5/sdf/sdf.hh:2,
                 from /usr/include/gazebo-11/gazebo/common/Plugin.hh:33,
                 from /homes/corcodel/ws_robotiq/src/robotiq/robotiq_3f_gripper_articulated_gazebo_plugins/include/robotiq_3f_gripper_articulated_gazebo_plugins/RobotiqHandPlugin.h:34,
                 from /homes/corcodel/ws_robotiq/src/robotiq/robotiq_3f_gripper_articulated_gazebo_plugins/src/RobotiqHandPlugin.cpp:38:
/usr/include/ignition/math6/ignition/math/Helpers.hh:40:13: warning: variable templates only available with ‘-std=c++14’ or ‘-std=gnu++14’
   40 | constexpr T IGN_MASSMATRIX3_DEFAULT_TOLERANCE = T(10);
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/ignition/math6/ignition/math/Helpers.hh: In function ‘std::chrono::_V2::steady_clock::duration ignition::math::v6::stringToDuration(const string&)’:
/usr/include/ignition/math6/ignition/math/Helpers.hh:990:28: error: ‘chrono_literals’ is not a namespace-name
  990 |       using namespace std::chrono_literals;
      |                            ^~~~~~~~~~~~~~~
/usr/include/ignition/math6/ignition/math/Helpers.hh: In function ‘std::chrono::_V2::steady_clock::time_point ignition::math::v6::stringToTimePoint(const string&)’:
/usr/include/ignition/math6/ignition/math/Helpers.hh:1032:28: error: ‘chrono_literals’ is not a namespace-name
 1032 |       using namespace std::chrono_literals;
```
I have modified ```add_compile_options(-std=c++11)``` to ```add_compile_options(-std=c++17)``` in file robotiq_3f_gripper_articulated_gazebo_plugins/CMakeLists.txt then rebuilt. The old errors are gone but not I'm getting this error and frankly I'm not sure what to make of it:

```
In file included from /usr/include/sdformat-9.5/sdf/Element.hh:20,
                 from /usr/include/sdformat-9.5/sdf/Actor.hh:25,
                 from /usr/include/sdformat-9.5/sdf/sdf.hh:2,
                 from /usr/include/gazebo-11/gazebo/common/Plugin.hh:33,
                 from /homes/corcodel/ws_robotiq/src/robotiq/robotiq_3f_gripper_articulated_gazebo_plugins/include/robotiq_3f_gripper_articulated_gazebo_plugins/RobotiqHandPlugin.h:34,
                 from /homes/corcodel/ws_robotiq/src/robotiq/robotiq_3f_gripper_articulated_gazebo_plugins/src/RobotiqHandPlugin.cpp:38:
/usr/include/c++/9/any:380:7: error: ‘struct std::any::_Manager_internal<_Tp>’ redeclared with different access
  380 |       struct _Manager_internal
      |       ^~~~~~
/usr/include/c++/9/any:404:7: error: ‘struct std::any::_Manager_external<_Tp>’ redeclared with different access
  404 |       struct _Manager_external
      |       ^~~~~~
```

It's clearly a dependency to Gazebo 11 (on Noetic) but the plugins definition looks fine, I don't understand what it complaining about. Any ideas would be appreciated.

Best,
Radu

RaduCorcodel, 2021-07-09:
I poked around more and I localized it to [Line34-39](https://github.com/ros-industrial/robotiq/blob/45196f6558fe8ba9d89bc8a105396c68c3e7e892/robotiq_3f_gripper_articulated_gazebo_plugins/src/RobotiqHandPlugin.cpp#L34). Which would make sense because the compiler complains about access.

nhinke, 2021-09-07:
Hi @RaduCorcodel ,

I am currently experiencing the same issue that you outlined above; were you ever able to figure out a solution? Thanks!

mqt0029, 2021-09-16:
Have a look at [jr-robotics/robotiq](https://github.com/jr-robotics/robotiq), which does have a `noetic-devel` branch currently in the works. Apparently, the fix was removing the `#define private public` and renamed a few references. See 7a8f61b26325ab01002e0c7ebd1aa53ab5c25317.

adamanov, 2022-06-30:
There is also an alternative repo which in maintained by University Hamburg 
https://github.com/TAMS-Group/robotiq