ros-industrial-attic/robotiq #195: "rosdep install robotiq_modbus_tcp" failed to execute

MZJR, 2021-11-30:
- Version:
Ubuntu20.04 noetic
- Connection method:
Robotiq_85gripper plug directly into the flange interface at the end of the UR5e,PC and UR control box are connected by network cable

I don't know if it is because of Noetic. If so, how can I control the robotiq gripper connected to UR under this version?

The following error appears as the title:
`ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install python-pymodbus] failed`

srediske, 2021-11-30:
Rosdep can't find a reference in their config file ([should be this one](https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml)) for `python-pymodbus` so it tries to call it directly, which fails.

You could try `apt-get install python3-pymodbus` or follow the instructions: [pymodbus](https://pymodbus.readthedocs.io/en/latest/readme.html#installing). But there could be more issues with noetic on this old repo.

MZJR, 2021-12-01:
> Rosdep can't find a reference in their config file ([should be this one](https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml)) for `python-pymodbus` so it tries to call it directly, which [...]

Thanks for your suggestion. It works.

closed 2021-12-01 by MZJR