ros-industrial-attic/robotiq #200: Can I maintain this package?

lianghongzhuo, 2022-05-09:
Can I maintain this package?

nicolaslauzier, 2022-05-10:
Sure! Do you have access to the Robotiq devices?

-- 

Nicolas Lauzier, Eng., Ph. D. / ing.

Senior Product Manager⼁Chef de produit senior

1-888-ROBOTIQ (762-6847)

1-418-380-2788 (Outside US & Canada)

<https://robotiq.com/>

  <https://www.linkedin.com/company/robotiq/>
<https://www.facebook.com/robotiq>  <https://www.instagram.com/robotiqinc>
<https://twitter.com/Robotiq_Inc>
<https://www.youtube.com/channel/UCHpdi06hmy0SSjrGqfmLZqQ>

lianghongzhuo, 2022-05-10:
I have access of the robotiq 3finger gripper with force torque sensor. And we have a custom Bluetooth adapter

nicolaslauzier, 2022-05-10:
> @nicolaslauzier <https://github.com/nicolaslauzier>: would this be an
> opportunity to discuss maintenance of this repository in general?
>
> It's been a while since you've been active here, but it appears you're
> still monitoring.

Robotiq has stopped the support of the packages as we are more focussed on
our other types of integration with robots, but if community members want
to support the packages that would be perfect.

-- 

Nicolas Lauzier, Eng., Ph. D. / ing.

Senior Product Manager⼁Chef de produit senior

1-888-ROBOTIQ (762-6847)

1-418-380-2788 (Outside US & Canada)

<https://robotiq.com/>

  <https://www.linkedin.com/company/robotiq/>
<https://www.facebook.com/robotiq>  <https://www.instagram.com/robotiqinc>
<https://twitter.com/Robotiq_Inc>
<https://www.youtube.com/channel/UCHpdi06hmy0SSjrGqfmLZqQ>

gavanderhoorn, 2022-05-10:
If that's the case, I'd like to suggest migrating this repository out of the `ros-industrial` organisation.

There is nothing linking the packages here to the ros-industrial effort (any more), and hosting the repository here gives off a wrong impression.

gavanderhoorn, 2022-05-10:
> Robotiq has stopped the support of the packages as we are more focussed on
our other types of integration with robots

an official statement -- added to the readme for instance -- about this would also have been appreciated.

It would have at least informed your customers what the status of the repository and the packages in it is.

Leaving everything in limbo by abandoning it has led to a great deal of fragmentation, duplication of effort and alternative packages.

Very unfortunate.

nicolaslauzier, 2022-05-10:
Sorry, my answer was not clear.

We are not actively maintaining and developing those packages but we are
helping our customers as much as possible if they experience issues. We
have worked with external resources for a certain time but it is not
something we've been able to do recently.

I'll look to clarify the status of the package.

-- 

Nicolas Lauzier, Eng., Ph. D. / ing.

Senior Product Manager⼁Chef de produit senior

1-888-ROBOTIQ (762-6847)

1-418-380-2788 (Outside US & Canada)

<https://robotiq.com/>

  <https://www.linkedin.com/company/robotiq/>
<https://www.facebook.com/robotiq>  <https://www.instagram.com/robotiqinc>
<https://twitter.com/Robotiq_Inc>
<https://www.youtube.com/channel/UCHpdi06hmy0SSjrGqfmLZqQ>

lianghongzhuo, 2022-05-11:
I'm currently maintaining our robotiq gripper. Feel free to use our fork which supports ros noetic.
https://github.com/TAMS-Group/robotiq
@gavanderhoorn feel free to link our repo as your repo is outdated.