ros-industrial-attic/robotiq #201: How to speed up the 'gripper.sendCommand()' process?

KentoYamamotoNU, 2022-07-04:
I run the 'robotiq_2f_gripper_control/nodes/Robotiq2FGripperRtuNode.py', and found that the frequency of the published topic 'Robotiq2FGripper_robot_input' is about 2.193hz. I would like to speed up the frame rate to at least 60hz. I think 'gripper.sendCommand()' is taking a long time to process. Anyone knows how to speed up the 'gripper.sendCommand()' process?

The following is the results of the commond 'rostopic hz /Robotiq2FGripper_robot_input'.

average rate: 2.193
	min: 0.456s max: 0.456s std dev: 0.00000s window: 2
average rate: 2.193
	min: 0.456s max: 0.456s std dev: 0.00005s window: 4
average rate: 2.194
	min: 0.455s max: 0.456s std dev: 0.00043s window: 6
average rate: 2.194
	min: 0.455s max: 0.456s std dev: 0.00036s window: 9
average rate: 2.194
	min: 0.455s max: 0.456s std dev: 0.00040s window: 11
average rate: 2.194
	min: 0.455s max: 0.456s std dev: 0.00038s window: 13
average rate: 2.193
	min: 0.455s max: 0.457s std dev: 0.00044s window: 15
average rate: 2.194
	min: 0.455s max: 0.457s std dev: 0.00047s window: 17
^Caverage rate: 2.194
	min: 0.455s max: 0.457s std dev: 0.00046s window: 18

KentoYamamotoNU, 2022-07-08:
I found solution. 
Thank you.

closed 2022-07-08 by KentoYamamotoNU

gavanderhoorn, 2022-07-08:
You might want to explain what you did/changed, to help your fellow Robotiq users.