ros-industrial-attic/robotiq #207: ROS2 - Robotiq Force Torque Sensor - Ready

panagelak, 2023-07-01:
Hello to whoever is interested i have migrated the Robotiq ft sensor driver to ros2 (tested in humble)

* As a standalone node or a composable ros2 control hardware interface 
* Additional feature include "fake/pass through mode" to test without hardware
* And an optional subscriber to add force to the measurements from external node

have a look : https://github.com/panagelak/robotiq_ft_sensor_driver

I tried to keep this repo maintainers in the package.xml and the license

Enjoy.

panagelak, 2024-02-07:
hi you can send me an email at panagiotis.angelakis.robot@gmail.com until i re-upload it

uartan, 2024-02-07:
@panagelak thank you very much works great!

:)